lgsvl/simulator #222: support no render mode docker build

david-gwa, 2019-07-01:
hey, lg-team,

     there is a [docker build](https://github.com/lgsvl/simulator/tree/master/Docker) in lg-sim, as I understand this is only for build,  not to run lg-sm in docker env, right ?

    will lg-team consider build a render-less docker file, so can build and run in one docker env ?

   thanks
  David

EricBoiseLGSVL, 2019-07-08:
@david-gwa 

The docker is for simplifying team builds during development.  It is the same as our build instructions seen here: https://github.com/lgsvl/simulator/blob/master/Docs/docs/build-instructions.md

To build and run headless in a docker env is not supported currently.  We have a headless mode in our new HDRP Simulator but we don't have plans to include it in a docker env yet.  If this is something that would be helpful, we can discuss adding it to our roadmap.

martins-mozeiko, 2019-07-12:
@david-gwa running simulator in docker container should be pretty straight forward, it does not require anything special. If you have Nvidia GPU then I suggest you take [nvidia/opengl](https://hub.docker.com/r/nvidia/opengl) container as base. Install Unity linux depenendencies ([from here](https://forum.unity.com/threads/unity-on-linux-release-notes-and-known-issues.350256/)), then put simulator in and it should work.

david-gwa, 2019-07-13:
thanks, @Martina, and @Eric 

 [docker build](https://github.com/lgsvl/simulator/tree/master/Docker) has these dependencies already, so I just used it. while  running  `docker-compose build`,  it was blocked as following: 

[image]

it looks like a unity relate issue.

martins-mozeiko, 2019-07-14:
This is normal message and it is displayed while Unity is being downloaded and installed (UnitySetup executable). Check your network and CPU usage - you'll see that Unity is being download first, and then installed. Depending on your network and computer speed this can take from few minutes to 10 or 20 min.

But you do not need to do this for running simulator. Unity inside docker is needed only for building simulator. You can completely skip running UnitySetup executable to just run simulator.

david-gwa, 2019-07-15:
hey, @martins-mozeiko , thanks for pointing out.

I tried today with the similar Dockerfile for running simulator:

```xml 
FROM ubuntu:16.04 

RUN apt-get update   \
    && apt-get install -y --no-install-recommends \
    && python3 python3-pip \
    && alsa-utils \ 
    && libcanberra-gtk-module libcanberra-gtk3-module
    &&    rm -rf /var/lib/apt/list/* 

RUN set -ex \
    && apt-get update \
    && DEBIAN_FRONTEND=noninteractive apt-get upgrade -y \
    && DEBIAN_FRONTEND=noninteractive apt-get install -y \
        sudo curl xvfb zip \
        gconf-service lib32gcc1 lib32stdc++6 libasound2 libc6 libc6-i386 libcairo2 libcap2 libcups2 \
        libdbus-1-3 libexpat1 libfontconfig1 libfreetype6 libgcc1 libgconf-2-4 libgdk-pixbuf2.0-0 \
        libgl1 libglib2.0-0 libglu1 libgtk2.0-0 libgtk-3-0 libnspr4 libnss3 libpango1.0-0 libstdc++6 \
        libx11-6 libxcomposite1 libxcursor1 libxdamage1 libxext6 libxfixes3 libxi6 libxrandr2 \
        libxrender1 libxtst6 zlib1g debconf npm libgtk2.0-0 libsoup2.4-1 libarchive13 libpng16-16 \
    && apt-get clean 

# nvidia runtime
COPY --from=nvidia/opengl:1.1-glvnd-runtime-ubuntu16.04 \
 /usr/local/lib/x86_64-linux-gnu \
 /usr/local/lib/x86_64-linux-gnu

COPY --from=nvidia/opengl:1.1-glvnd-runtime-ubuntu16.04 \
 /usr/local/share/glvnd/egl_vendor.d/10_nvidia.json \
 /usr/local/share/glvnd/egl_vendor.d/10_nvidia.json

 RUN echo '/usr/local/lib/x86_64-linux-gnu' >> /etc/ld.so.conf.d/glvnd.conf && \
 ldconfig && \
 echo '/usr/local/$LIB/libGL.so.1' >> /etc/ld.so.preload && \
 echo '/usr/local/$LIB/libEGL.so.1' >> /etc/ld.so.preload

# nvidia-container-runtime
ENV NVIDIA_VISIBLE_DEVICES ${NVIDIA_VISIBLE_DEVICES:-all}
ENV NVIDIA_DRIVER_CAPABILITIES ${NVIDIA_DRIVER_CAPABILITIES:+$NVIDIA_DRIVER_CAPABILITIES,}graphics

#COPY ./lgsvl-api/lgsvl-0.0.0-py3.whl /tmp/lgsvl-0.0.0-py3.whl

#RUN pip install /tmp/lgsvl-0.0.0-py3.whl && \
#    rm /tmp/lgsvl-0.0.0-py3.whl

WORKDIR /root

``` 

this Dockerfile is  workable to build the runtime simulator image, then attach the simulator-binary built on host to the container, it starts all right and jump out the `Automotive Simulator Configuration` window

after I selected a certain  screen resolution, it crashed: 

### crash info 

root@ubuntu:/lg_bin# ./simulator 
Set current directory to /lg_bin
Found path: /lg_bin/simulator
Mono path[0] = '/lg_bin/simulator_Data/Managed'
Mono config path = '/lg_bin/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'libAsyncTextureReader.so'
Preloaded 'libpng.so'
Preloaded 'libturbojpeg.so'
Preloaded 'ScreenSelector.so'
Display 0 'HP Z22n G2 22"': 1920x1080 (primary device).
Loading player data from /lg_bin/simulator_Data/data.unity3d
[... 517 lines cut ...]

	/lg_bin/simulator_Data/MonoBleedingEdge/x86_64/libmonobdwgc-2.0.so(+0xd42ce) [0x7fd34bba32ce]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7fd3541f5390]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x38) [0x7fd3533a6428]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x16a) [0x7fd3533a802a]
	./simulator() [0xb4610c]
	/lg_bin/simulator_Data/MonoBleedingEdge/x86_64/libmonobdwgc-2.0.so(+0x14d876) [0x7fd34bc1c876]
	/lg_bin/simulator_Data/MonoBleedingEdge/x86_64/libmonobdwgc-2.0.so(+0x5af41) [0x7fd34bb29f41]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7fd3541f5390]

Debug info from gdb:

mono_gdb_render_native_backtraces not supported on this platform, unable to find gdb or lldb

=================================================================
Got a SIGABRT while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

Aborted (core dumped)

Is there a suggestion here ? 

Thanks again

martins-mozeiko, 2019-07-15:
What docker command you are using to run this docker image?
Make sure you are using nvidia runtime and are passing all necessary X11 stuff (unix domain socket, XAUTHORITY and DISPLAY env variables). You can see example of what things you need to use from same second-ros repository in [docker-compose.yml](https://github.com/lgsvl/second-ros/blob/master/docker-compose.yml#L14) file

Easy way to check if GPU will work is to run `glxinfo -B` command inside docker container. It is available in `mesa-utils` package (apt get install will install it for you). If container runs correctly, you'll see output similar to this:

```
# glxinfo -B
name of display: :0
display: :0  screen: 0
direct rendering: Yes
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1080/PCIe/SSE2
OpenGL core profile version string: 4.5.0 NVIDIA 430.26
OpenGL core profile shading language version string: 4.50 NVIDIA
OpenGL core profile context flags: (none)  
OpenGL core profile profile mask: core profile
                                                                                   
OpenGL version string: 4.6.0 NVIDIA 430.26
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
                                                                                   
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 430.26
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

Important parts here is having `Yes` for `direct rendering` and having your GPU name visible in `OpenGL renderer string`.

Make sure glxinfo works before running simulator.

david-gwa, 2019-07-16:
thanks again, it works now: 

[image]

closed 2019-08-20 by martins-mozeiko