lgsvl/simulator #223: Failed to send a routing request in new map

dongyy1148, 2019-07-01:
Hi,

I want to try the map replacement function in your simulator, so I removed all the buildings in your "SimpleMap" map and then replace the old "base_map.bin" file and "generate_map.sh simple_map". And then I can see my new map in Unity editor and I connect it with Apollo 3.5 to do the test. However, when I submit the target point in Apollo, it returns that "Failed to send a routing request in new map" and there is no planned route showed (no red line) and the vehicle doesn't move. There is always a purple obstacle in front of the vehicle and its size keeps changing showed in figure. When I move the vehicle by keyboard, the purple thing will also move. However, if I change the map to "San Francisco", there is no problem showed. Could you help give some advice? Thanks a lot!
[image]

EricBoiseLGSVL, 2019-07-08:
@dongyy1148 

Try to set a point of interest on the same lane of the vehicle, close to it and see if it still fails.

Why are you needing to replace the current .bin and .sh?  If you are adding annotations and they are not correct, the new files maybe broken.  You should debug inside Apollo to tell what the issue is.

dongyy1148, 2019-07-09:
@EricBoiseLGSVL If I want to replace the map file in Apollo, I should use HDMap tool in unity to generate a new bin file and replace the old one right? The use the .sh to generate the suitable map which can be used by APollo (I don't replace .sh file). Besides, I have tried to add a very close point, but it still doesn't work.

luqiang21, 2019-07-09:
@dongyy1148 Yes, you're right. If you want to replace the map file, you need to generate a new `bin` file and run `.sh` file inside apollo docker. We are not sure why the routing doesn't work. You can try to debug it inside apollo by running routing in a terminal instead of clicking the button in dreamview. The terminal will have some useful debug messages.

closed 2020-04-07 by EricBoiseLGSVL