lgsvl/simulator #225: New map shows in Apollo but not in LGSVL

dongyy1148, 2019-07-01:
Hi LGSVL team,

I have made new vehicle and new map in Unity, replace the map file under APollo map folder and also build the entire LGSVL project to update the simulator. Now my new truck shows in the LGSVL, my new map shows in the dreamview, however, my new map doesn't show in the LGSVL simulator's choose panel. I can see my new map in the Unity editor but not in LGSVL simulator. Could you give some advice? Thanks!

daohu527, 2019-07-05:
You can try "run command" in Unity, check if you can see the map in the game window. Or confirm that the coordinates are correct?

dongyy1148, 2019-07-05:
@daohu527 I can see my new map in the game window and also Apollo dreamview, but not for the LGSVL simulator.

daohu527, 2019-07-06:
Now, I don't have a good suggestion. I will try to reproduce this question and will give you reply next Tuesday.

dongyy1148, 2019-07-08:
@daohu527 Thanks a lot! I really appreciate your help!

EricBoiseLGSVL, 2019-07-08:
@dongyy1148 

Hello, let's see if I can help with this issue.

1.  Select Assets/ScriptableObjects/DefaultAssetBundleSettings.asset
2.  Increase the Maps array count
3.  Drag the new scene into the Scene Asset field
4.  Create a screen shot in editor of the new map (.png) and place in Assets/Textures/
5.  Make sure the map image is named... map_yourmap.png and the map is Yourmap.scene
6.  Drag the map_yourmap.png into the SpriteImg field.
7.  Save scene so the asset updates in source control

In editor (I think you have this already if it is working in editor but just in case)
1.  File -> Build Settings
2.  Drag the Yourmap.scene into the list

Hope this helps.

daohu527, 2019-07-10:
I try the whole process, and it seems work. a small supplement.
1. Select Assets/ScriptableObjects/DefaultAssetBundleSettings.asset
2. Increase the Maps array count
3. Drag the new scene into the Scene Asset field
4. Create a screen shot in editor of the new map (.png) and place in Assets/Textures/
5. Make sure the map image is named... map_yourmap.png and the map is Yourmap.scene, **Set the "Texture Type" of yourmap.png to "Sprite (2D and UI)" in the Inspector panel.**
6. Drag the map_yourmap.png into the SpriteImg field.
7. Save scene so the asset updates in source control

closed 2020-04-07 by EricBoiseLGSVL