lgsvl/simulator #226: Publish traffic lights information

n1try, 2019-07-02:
Is there any plans to implement publishing of ground truth traffic lights information, i.e. position and state (red, yellow, green) to Rosbridge?

daviduhm, 2019-07-03:
Yes, we are planning to support this feature that will publish ground truth information for traffic lights. At the moment, we are preparing to release a new version of our simulator and all of new features will most likely be added to the new one, which will come later. Please stay tuned!

ninjacoder0, 2019-07-03:
Do you have any rough dates on new releases?

daviduhm, 2019-07-03:
We are targeting to release by the *end of July*, but that could possibly be changed. We'll officially announce the exact release date later on soon.

closed 2019-07-04 by n1try