lgsvl/simulator #229: what information does the radar sesor create?

alexsimon1, 2019-07-04:
This issue is because the XE_Rigged-lgsvl vehicle has a radar sensor with incorrect target environment, *Apollo*. At the moment, our radar sensor implementation only supports Apollo, which means it creates a ros/cyber message in apollo specific topic type `apollo.drivers.ContiRadarObs` and publishes to rosbridge. You are getting the error message because most probably you are not running Apollo with XE_Rigged-lgsvl vehicle and thus your rosbridge doesn't know about this message type.

I guess we should better remove a radar sensor from XE_Rigged-lgsvl vehicle as it doesn't support target environments other than Apollo. Supporting radar sensor for environments other than Apollo is in our plan, but that will come later. Alternatively, you can write your own message type definitions and code to publish it. Please take a look at [RadarSensor.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/RadarSensor.cs) and [PbMsgs.cs](https://github.com/lgsvl/simulator/blob/91d60e9232ebfc2b3fc013265af8ef02c1d907b0/Assets/Scripts/Ros/PbMsgs.cs#L788). You can see there how we define message definitions for radar and how to create a message and publish it to rosbridge.

_Originally posted by @daviduhm in https://github.com/lgsvl/simulator/issues/227#issuecomment-508272910_

hello, 
i'm not fully understand what Information the Radar sensor create? Does the Radar create a PointCloud?

daviduhm, 2019-07-05:
No, only the Lidar sensor creates point clouds. Radar sensor detects obstacles when an obstacle enters into the Radar detection range and creates a ContiRadar message about that obstacle.

From the [RadarSensor.cs#L324](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/RadarSensor.cs#L324), you can see that we are creating a ContiRadar message, which has a list named ContiRadarObs for every detected obstacles. This message type is for Apollo specific, and Apollo expects a radar message to have `apollo.drivers.ContiRadar` topic type with those bunch of fields (e.g., obstacle ID, a position of an obstacle, dimensions of an obstacle, etc).

For the detailed information about what each field means, you can take a look at [conti_radar.proto](https://github.com/ApolloAuto/apollo/blob/master/modules/drivers/proto/conti_radar.proto) from Apollo repo. This is the protobuf message definition we referenced for implementing our Radar sensor.

alexsimon1, 2019-07-05:
ok, so the radar in the Simulator don't work like a normal radar sensor in a real car? Because usually a radar creates points with the distance. 

Does your Radar create points and distances in the background and could I get them with a modification of your script? 

Or does your Radar work more like a camera and there are no points at all?

I thank you in advance for your help :)

daviduhm, 2019-07-05:
I'm not sure if I fully understand your question, but the radar *does have* a point information on the obstacle (e.g., an NPC vehicle) it's currently detecting. A point can be expressed as a vector, and if you check the ContiRadarObs message definition from [conti_radar.proto#L30](https://github.com/ApolloAuto/apollo/blob/05412325a74c09d9438fa4533a6c58542c6caab6/modules/drivers/proto/conti_radar.proto#L30), it has a `longitude_dist` field and a `lateral_dist` field: `longitude_dist` refers to longitude distance between the radar and an obstacle and `lateral_dist` refers to lateral distance between the radar and an obstacle. If you want to get those distance information, you can subscribe to the topic `/apollo/sensor/conti_radar` using Apollo.

>                 x axis  ^
>                         | longitude_dist
>                         |
>                         |
>                         |
>           lateral_dist  |
>           y axis        |
>         <----------------
>         ooooooooooooo   //radar front surface

*Quoted from [conti_radar.proto#L30](https://github.com/ApolloAuto/apollo/blob/05412325a74c09d9438fa4533a6c58542c6caab6/modules/drivers/proto/conti_radar.proto#L30)*

The picture below shows a raycast shot from the radar to the obstacles in radar detection ranges. You can think of the yellow line as a vector starting from the radar sensor and the tip of this vector is the point where the radar is currently hitting to.

[image]

alexsimon1, 2019-07-06:
I mean something like in the picture below. There you can see the Matlab Driving Scenario Designer simulate a typical radar and the point cloud its generating.

And if I understand you correct, LGSVL creates one point per obstacle (only vehicles?) its detects. 

[image]

daviduhm, 2019-07-08:
You are right. Currently, our radar creates one point per obstacle for vehicles *as well as* pedestrians in the scene. As in #227, [this radar sensor](https://github.com/ApolloAuto/apollo/blob/master/docs/specs/Radar/Continental_ARS408-21_Radar_Installation_Guide.md) is designed for supporting Apollo - which expects only one point per obstacle.

alexsimon1, 2019-07-08:
Thank you very much for your help

alexsimon1, 2019-07-09:
Another Question. 
If I wanted  to implement  any of the Sensors in this list: http://wiki.ros.org/Sensors
Do i have to modify the Scripts? 
For example this Lidar: https://www.sick.com/de/de/mess-und-detektionsloesungen/3d-lidar-sensoren/mrs6000/mrs6124r-131001/p/p533545

I probably  need to change the Lidar Script, because this Lidar don’t messeaure in 360 degrees?

daviduhm, 2019-07-09:
Sure it is possible to implement any of those sensors in the list. The way how you implement it is really depends on which sensor you are developing. You can modify our existing source code or you can also implement it from the scratch. In either cases, you'd need to have some knowledge/experience in C# and Unity field though.

As for the Lidar, you can see how we implemented it in [LidarSensor.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/LidarSensor/LidarSensor.cs). If you want to change the rotation angle of Lidar, I'd suggest you to take a look at [LidarSensor.cs#L290](https://github.com/lgsvl/simulator/blob/538a221983633fa94cbe5a2cedb45f58f88cc5e2/Assets/Scripts/LidarSensor/LidarSensor.cs#L290) and start from there by adjusting it per your requirements. Below I adjusted the Lidar to rotate only 180 degrees in front (this was just a quick fix and not tested thoroughly):
[image]

If you have more questions about other sensors such as Lidar, let me kindly suggest you to open up a new thread as it's off topic and not related to the Radar sensor. Thanks.

closed 2019-07-10 by alexsimon1