lgsvl/simulator #230: lgsvl_msgs submodule cannot be checkout for latest lgsvl apollo-3.5

Fibonacci43, 2019-07-05:
Cloning of the latest lgsvl apollo-3.5 repo gives following error:
I got "Unable to check out " for the git clone
[image]
And later, I tried "git submodule update", it still unsolved.Could you give me some advise?
[image]

daviduhm, 2019-07-05:
Ah, sorry about the inconvenience. We recently updated a submodule commit ID for lgsvl_msgs from apollo 3.5 repo but the latest commit to lgsvl_msgs repo hasn't been pushed. That's why it's unable to check out. I've just pushed the [commit](https://github.com/lgsvl/lgsvl_msgs/commit/af755bd67452916b97a6569abee4a24bddbf394e) to Github, and it should be fixed now. Could you try again please?

Fibonacci43, 2019-07-05:
> Ah, sorry about the inconvenience. We recently updated a submodule commit ID for lgsvl_msgs from apollo 3.5 repo but the latest commit to lgsvl_msgs repo hasn't been pushed. That's why it's unable [...]

Thank you for your operation, I get the right check out.

pengpingliang, 2019-07-08:
@daviduhm 

Are the lgsvl_msgs in use ? I try to find out where the messages be used in apollo-3.5 source, but I failed.

Could you tell me how the messages work in source ?  Thank you!

daviduhm, 2019-07-08:
lgsvl_msgs is the custom ROS message definitions to subscribe *Ground truth messages* being published by the LGSVL Simulator. It's our simulator feature and is implemented/being used in [GroundTruthSensor2D.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/GroundTruthSensor2D.cs) and [GroundTruthSensor3D.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/GroundTruthSensor3D.cs). Basically, the LGSVL Simulator publishes ground truth obstacle information to rosbridge such as a position, rotation, and dimensions of an obstacle. For details, you can check our documentation for ground truth obstacles: [https://www.lgsvlsimulator.com/docs/perception-ground-truth](https://www.lgsvlsimulator.com/docs/perception-ground-truth)

pengpingliang, 2019-07-09:
@daviduhm 
Thank you very much! I got it.

closed 2019-07-09 by Fibonacci43