lgsvl/simulator #231: Autoware setfacl operation not permitted

vatna, 2019-07-05:
I seem to have built the Autoware docker image ok but when I try to run it, I get the error: `setfacl: /home/e/.Xauthority: Operation not permitted`. Any ideas as to what I should do to correct this?
[image]

daviduhm, 2019-07-09:
Which Ubuntu version are you running on your host machine? Can you run `ll /home/e/.Xauthority` and see if this file exists and your home user (i.e., e) have an ownership for this file? Can you also do `echo $XAUTHORITY` and paste the output here?

vatna, 2019-07-09:
I'm running Ubuntu 18.04.2 LTS. Please see screenshot below.
[image]

daviduhm, 2019-07-09:
This issue is because inside [run.sh](https://github.com/lgsvl/Autoware/blob/lgsvl_develop/docker/generic/run.sh) in Autoware it's assuming that your Xauthority file is located in your home directory (`/home/$USER/.Xauthority), but apparently it's not in your case. Usually your $XAUTHORITY env variable should tell you where Xauthority file is, but it's also missing somehow. You can then run xauth in command line. It will tell you which authority file it's currently using.

In my Ubuntu 18.04:
```
$ echo $XAUTHORITY
/run/user/1000/gdm/Xauthority
```
```
$ xauth
Using authority file /run/user/1000/gdm/Xauthority
xauth>
```

Once you found your Xauthority file path, you can update [run.sh#L4](https://github.com/lgsvl/Autoware/blob/4fdb2f8d1aa87ca28f1f4f6a47bf2bf27951a7d5/docker/generic/run.sh#L4) to `XAUTH=/path/to/Xauthority`. Then, please try `bash run.sh kinetic` again and let me know how that works out for you.

vatna, 2019-07-10:
Mine seems to be in the same location I think. I notice that .Xauthority is owned by root and not autoware. is this how it should be? Please have a look at the screenshot.
[image]

daviduhm, 2019-07-10:
Ah, I see. You are supposed to run `echo $XAUTHORITY` on your host machine, not inside the Autoware docker container. Just update your [run.sh#L4](https://github.com/lgsvl/Autoware/blob/4fdb2f8d1aa87ca28f1f4f6a47bf2bf27951a7d5/docker/generic/run.sh#L4) to `XAUTH=$XAUTHORITY` and try again. Then, inside the docker, it should look as below. Note that .Xauthority is a file (not a directory) and owned by autoware:
```
$ ls -al /home/autoware
total 56
drwxr-xr-x  1 autoware autoware 4096 Jul 10 21:40 .
drwxr-xr-x  1 root     root     4096 May 30 00:34 ..
drwxr-xr-x  1 autoware root     4096 May 30 00:43 autoware
-rw-r--r--  1 autoware autoware  220 Aug 31  2015 .bash_logout
-rw-r--r--  1 autoware autoware 3948 May 30 00:51 .bashrc
drwxr-xr-x  3 autoware autoware 4096 May 30 00:45 .cmake
drwxr-xr-x  1 autoware autoware 4096 May 30 00:34 darknet
drwx------  1 autoware autoware 4096 May 30 00:51 .gconf
-rw-r--r--  1 autoware autoware  655 May 16  2017 .profile
drwxr-xr-x  3 autoware autoware 4096 May 30 00:34 .ros
drwxr-xr-x  3 autoware autoware 4096 May 29 23:36 shared_dir
drwxr-xr-x  1 autoware autoware 4096 May 30 00:35 ssdcaffe
-rw-r--r--  1 autoware autoware    0 May 30 00:34 .sudo_as_admin_successful
-rwxr-----+ 1 autoware autoware  114 Jul  1 18:03 .Xauthority
```

vatna, 2019-07-12:
In the screenshot above, I ran `echo $XAUTHORITY` on both the host and inside the docker container. But anyhow I made the change to run.sh as you suggested and it is now working. Thank you very much for your help!!!

closed 2019-07-12 by vatna