lgsvl/simulator #232: Apollo 3.5 fails to run with LGSVL Simulator

Fibonacci43, 2019-07-09:
Hello!
I try my best to run apollo 3.5 with lgsvl simulator, but struck into the final problem:get the car to run.
I followed [this instruction](https://github.com/lgsvl/apollo-3.5) , there is a [video](https://youtu.be/SmCvI-0U9kQ) to illustrate the running steps and the problem. 
It seems that the routing process are successed. But some other modules such as perception, planning and control are not start up. Is there a connection between this problem and the warning information when I type "./docker/scripts/dev_start.sh"? 

Thanks in advance for any help!

luqiang21, 2019-07-09:
@Fibonacci43 Thank you for your video. The process seems correct. Did you wait for some time before other modules like perception, prediction and planning to work? Typically, apollo 3.5 needs a bit time to start these modules. 

The warning/error information with `./docker/scripts/dev_start.sh` doesn't matter from our experience. You can refer more info [here](https://github.com/lgsvl/simulator/issues/195#issuecomment-506065585).

Fibonacci43, 2019-07-11:
Following your suggestion, I wait for at least five minutes, but nothing change. I tired many times, the problem still unsolved. I check the status of apollo modules. The perception module, prediction module, planning module are not running.  Also I found this issue (https://github.com/lgsvl/simulator/issues/184) describes the same problem that I have encountered. The owner solved it by reinstalling the nvidia-docker. I used nvidia docker 2.0 for now and I am sure that it is well installed and running properly in the docker,  I wonder what nvidia-docker version does this project use.  
By the way, do I have to install the cuda in my host computer to let the apollo perception run? I haven't installed cuda on my computer for now. Is there a connection between this and the problem?
Looking forward for your advice.

luqiang21, 2019-07-11:
@Fibonacci43 You don't need to install cuda because our docker image has it. Your nvidia docker version is the same as mine. Since your perception module is not working. Can you try to run it in a terminal instead of pressing the button. `perception.sh start_fe` should work. Then you can paste the error message here and we can look into it.

Fibonacci43, 2019-07-12:
@luqiang21 Hi, thank your for confirming my installation is correct. I have run the ' perception.sh start_fe '. And I got the information like below:
[image]
[image]
[image]

xmyqsh, 2019-07-12:
@Fibonacci43 The default extrinsic config is just an example. You should modify it to correspond to the LGSVL Simulator's setting of default vehicle in apollo-3.5, XE_Rigged_Apollo3.5. 
LGSVL has provoided all the material you need.

xmyqsh, 2019-07-12:
@luqiang21 
LGSVL's repo is an important supplementary of BAIDU Apollo!

martins-mozeiko, 2019-07-12:
@Fibonacci43 the error happens with cuda initialization - "check failed: error == cudaSuccess (8 vs. 0)"
What GPU are you using. Apollo oficially supports only GTX 1080.
If you are using supported GPU I suggest to reboot computer. Sometimes Nvidia cuda driver stops working and only way to fix it is to reboot computer.

Fibonacci43, 2019-07-15:
@martins-mozeiko  I tried the running steps again, it seems the main reason is actually the cuda initialization.  And I check the official apollo repository [related page](https://github.com/ApolloAuto/apollo/blob/master/docs/specs/Software_and_Kernel_Installation_guide.md), it suggests to use "375.39" version GPU driver. But my laptop GPU is GTX 1660Ti, which the suggested driver does not supprt. I will try to run on my another computer again , which have GTX 1070 intalled.

martins-mozeiko, 2019-07-15:
Yeah, 1660Ti is Tesla architecture GPU - meaning it is from newer generation than 1080. It is basically same category as RTX 2080. Which is not supported by Apollo. See ApolloAuto/apollo#4233 and ApolloAuto/apollo#6916 issues with more discussion on this.

closed 2020-04-07 by EricBoiseLGSVL