lgsvl/simulator #234: ROSAgentManager Not Found

Felixlxc, 2019-07-13:
Hi,
   I'm trying to add new vehicle in a new scene that I created for some tests. And I have followed the tutorial for add new ego-vehicle to add a player vehicle. However, when I run the simulation, I got some errors related to ROSAgentManger and the new vehicle (shown in the picture). Could you help me with this issue? Thanks a lot.
[image]
[image]

martins-mozeiko, 2019-07-14:
I think this is not related to how you added vehicle, but more about how new scene is created and used. You can try loading your vehicle in SanFrancisco scene to verify this.

How are you running new scene? Directly or through main menu? Are there any other debug log messages before these two?

Felixlxc, 2019-07-14:
@martins-mozeiko Thanks for the reply. I'm running the scene directly. I have tried to run it through main menu but I don't know how to adjust the initial position of the ego vehicle as the desired initial position. Is there a way to change the position of vehicle in the scene?

martins-mozeiko, 2019-07-14:
You control initial position by creating empty game object and attaching SpawnInfo script component. Vehicle will be created with transform of first game object that has SpawnInfo script (position & rotation).

Does this NullReferenceException error happens only once or is repeated many many times? If this happens only once, feel free to ignore it. It may be that startup order of objects currently is a bit wrong and VehicleController is doing something too early. You can check this by putting breakpoint in `SimulatorManager.cs` file, line 76, and checking that it is executed when you start scene directly. If it is, then just ignore this error message.

Felixlxc, 2019-07-14:
@martins-mozeiko 
The error only occur once for it and I'm able to use the keyboard to control the vehicle. Thanks for the information! It helps a lot!

Felixlxc, 2019-07-14:
@martins-mozeiko  I have tried to run in in the menu scene. There is no error about ROSAgentManager! So thank you very much!

closed 2019-07-14 by Felixlxc