lgsvl/simulator #235: Autoware and LGSVL starting

HartlD2, 2019-07-13:
I will always getting the following error if i press on ref of "Sensing" inside runtime manager:

['roslaunch', '/home/autoware/shared_dir/autoware-data/my_launch_sf_map/my_sensing_simulator.launch']
pid=410
sched policy=OTHER prio=0
ERROR: cannot launch node of type [rosbridge_server/rosbridge_websocket]: rosbridge_server
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/autoware/Autoware/ros/src
ROS path [2]=/opt/ros/kinetic/share

martins-mozeiko, 2019-07-16:
I think this is same issue as #131. Please try older commits of autoware-data repository to see if smaller map loads in rviz.

yangjiongfeng, 2019-07-30:
I made the same mistake as you.
How did you solve it?
Where can I download your autoware-data files?

martins-mozeiko, 2019-07-30:
Did you try 983372cbeb107ea1201509e36c commit? Autoware-data repository is here: https://github.com/lgsvl/autoware-data

yangjiongfeng, 2019-07-31:
where and How to check commit?

martins-mozeiko, 2019-07-31:
`git rev-parse HEAD` will show current commit. To get different commit use `git checkout COMMIT_ID` command.

closed 2020-04-07 by EricBoiseLGSVL