lgsvl/simulator #236: Running simulator on a virtual machine in the cloud?

vatna, 2019-07-14:
I was wondering if it would be possible to run the simulator together with Apollo 3.5 on a virtual machine, for example on Azure? I was looking at a machine such as a NC6 on https://azure.microsoft.com/en-gb/pricing/details/virtual-machines/linux/. My concern is that it has a K80 GPU and I don't know if everything will work with that?

martins-mozeiko, 2019-07-14:
Simulator will run with P100, V100 or M60. Not K80 - it doesn't virtualize display adapter that Unity will need to create rendering window. Only difference will be what kind of performance you will get. The newer GPU, the better performance.

I'm not sure exactly how GPU virtualization works in Azure, it may require installing special Nvidia vGPU driver so it can be used for rendering not only compute. See more information here: https://www.nvidia.com/en-us/data-center/graphics-cards-for-virtualization/ If Azure provides this out of the box, then you won't need to install anything extra.

Apollo will only run with GPU that is from same generation as GTX 1080. For example, Titan or RTX GPUs won't work. From your list most probably P100 will work as it is also Pascal architecture GPU - same as GTX 1080. But I cannot guarantee this - we have not tried this.

Technically newer generations of GPUs (V100) should also work, but that will require recompiling caffe library inside Apollo docker which is not an easy task - Apollo docker contains prebuilt binary for caffe library, and the build process is not really explained.

See ApolloAuto/apollo#4233 or ApolloAuto/apollo#6916 issues where are some discussions about Apollo supporting newer GPU architectures.

vatna, 2019-07-15:
Thanks for your detailed answer. I'll try to test this out on a P100 as you suggest.

closed 2019-07-15 by vatna