lgsvl/simulator #237: NPC error

Felixlxc, 2019-07-15:
Hi,  
 I'm using NPC feature in scene simulation in a newly created scene. And I have done map annotation in the scene. 
When I start to use "enable traffic", the error shown in the following figure occurs. Could you help me with this error?
[image]
[image]

EricBoiseLGSVL, 2019-07-15:
@Felixlxc Please check that IsSimplePhysics = true in NPCManager.cs attached to the NPCManager.prefab.  It can be found at Assets -> Prefabs -> NPCRefactor

This looks like you have complex physics enabled and it is not supported yet.  It is a feature in progress and has issues.

Felixlxc, 2019-07-15:
@EricBoiseLGSVL  Thanks for the help! I checked the IsSimplePhysics and this error still occurs. And I will get the error "NPCManager not found." I wonder if it is related to how I use the map annotation tool to generate lanes etc?

EricBoiseLGSVL, 2019-07-15:
@Felixlxc 

Hmm, The NpcManager should always be found.  Is the prefab listed in the SimulatorManager.prefab Managers list?

Can you screen shot your scene Hierarchy and the NPCManager prefab in Unity Editor please?

Felixlxc, 2019-07-15:
@EricBoiseLGSVL  Yeah sure.
[image]
This is my Hierarchy for the scene and Map_linden is where the map annotation lie in. The ADL_vehicle is added according to "add new ego vehicle" tutorial. 
[image]
And this is the NPCManager in inspector. Thank you for your help and I appreciate it a lot!

Felixlxc, 2019-07-16:
@EricBoiseLGSVL  And these are the related screenshot for the error. 
[image]
[image]
The first screenshot is the NPC Vehicle which cause the error and the second figure is about how I create lanes at such position. I would like to know if I make any mistake in building the lanes so that it causes the error "index out of range". Sorry for anther question here.

EricBoiseLGSVL, 2019-07-16:
@Felixlxc Thanks for the screen shots.  I will need to see a screen shot of the hierarchy when the scene is running. Please highlight the NPCManager prefab in the inspector, e.g.

[image]

I think you are correct that the annotation is missing something.  Let's look at that next.
So, we separate lane sections and intersections with parent gameobjects to make it easier to parse waypoints and connections.
Let's get a screen shot of the annotation hierarchy where another lane section is adjacent, e.g.

[image]

Note: MapLaneSection has a MapLaneSection class component and all MapLaneSegmentBuilders are child objects.  Also note that all first and last waypoints align with the previous and next first or last waypoints.  It doesn't need to be perfect but close.  It's hard to tell but it seems that you have a cross street and it is not annotated correctly because the NPC can't get the next lane.  Look at SimpleMap.scene to see a simplified map annotation with intersections and lane associations.

The error WaypointFetcher line: 90 is trying to get the next lane and getting a null.  The logic for isSimplePhysics is that the npc is pooled when the next lane MapLane is null.  I don't think there is a check in complex physics so the error is happening.

Oddly, this code should never be called if isSimplePhysics is == true;  This maybe a logic error and is getting called somehow.    I'll look into this more on our end.  For a test, find all isSimplePhysics checks and comment out the else (all complex physics code) in NPCControllerComponent.cs.
Also, comment out Line: 458 that checks if npc is in view of the ego vehicle.  This is for realism and will prevent despawn when missing next lane. 
If you see NPC's disappearing at the end of the road section constantly, then that is where you have an annotation issue. (I use this to check annotation issues internally :) )

closed 2019-07-21 by Felixlxc