lgsvl/simulator #238: Simulator crashed during scenario run

yanbec, 2019-07-15:
Hi!
I'm working on some automated scenarios in the simulator, currently using Apollo 3.5.
After the latest fixes it all runs a lot better and I was able to do some serious testing.
For this weekends' test runs, it seems the simulator crashed about ten hours in. I put the log (sadly, not the complete output) and all my relevant code into this [gist](https://gist.github.com/yanbec/b58c76474ddf3ef8da55f4ecc44e4afa).

The scenarioHelper module includes everything that will be the same for any scenario. I already used this for some other scenarios. 

Two things I struggled with in the beginning are already done:

- Performance issues in case you use the reset() function of the simluator. It's a lot better to manually delete the npc vehicle(s) and reuse the ego vehicle. There may be some memory leak or a similar bug.
- Apollo sometimes loses the destination point. Therefore, I manually send the destination for every run via websocket. I know this could be done in Python, but I preferred the use of a "[websocat](https://github.com/vi/websocat)"  system call for ease of use. This also enables automated test runs, as another scenario can just set it's own destination and can therebore be run without manual intervention.

All of this currently runs on a Ryzen 1600 with 32GiB memory, and a GTX1070 for the simulator and a GTX1050Ti for Apollo on Ubuntu 18.04.

Some kind of cheap workaround would be alright for me - **is it possible to start directly into the simulator without the setup window in the beginning?** This way I could just restart in case of a crash and continue testing.

Any help and/or ideas to resolve this or why this is happening is appreciated.

martins-mozeiko, 2019-07-15:
Unfortunately there is no way to disable this dialog with command-line arguments. But you can disable this dialog in project properties and then rebuild the binary which then won't show this dialog. Here are instructions how to disable it: https://stackoverflow.com/a/47109945

yanbec, 2019-07-16:
@martins-mozeiko thank you - until now I was unable to build the binary myself, but I'll have a try at this and report back. It crashed again last night (after just 74 test runs ..). I will try to catch some more logs, maybe it's possible to resolve this bug in the future.

yanbec, 2019-07-18:
Building the standalone executable following [these instructions](https://github.com/lgsvl/simulator/blob/master/Docs/docs/build-instructions.md) worked like a charm this time. Using an automatic restart script in combination with with some timeout additions and, if necessary, killing the simulator after timing out I was now able to do some nice test runs.  At this time, I can't give any more details to the crashes though.

martins-mozeiko, 2019-07-18:
When simulator crashes check the log file in `~/.config/unity3d/LG Silicon Valley Lab/Automotive Simulator/Player.log`  location. Maybe it will have useful information at the end.

closed 2020-04-07 by EricBoiseLGSVL