lgsvl/simulator #240: MPC Waypoints Follower in Autoware v1.12.0

josephqci, 2019-07-16:
Autoware 1.12.0 is released on July 08, 2019. It provides a new module - MPC Waypoints Follower. This module can make the car move forward and turn more smoothly. It needs vehicle status topic published from related node or simulator. Does lgsvl simulator plan to support this new function?

martins-mozeiko, 2019-07-16:
Yes, we are interested in supporting newer Autoware version. We'll look into supporting it.

closed 2020-04-07 by EricBoiseLGSVL

ozzdemir, 2020-12-02:
Any update on this issue? Is there a timeline for the implementation of vehicle_status sensor?

EricBoiseLGSVL, 2020-12-02:
We currently support 1.14.  Is the vehicle status topic supported in this version?  You can make a sensor plugin that publishes this data now.

mengen100, 2021-06-29:
> We currently support 1.14. Is the vehicle status topic supported in this version? You can make a sensor plugin that publishes this data now.

Hi Eric,
May I know which sensor supports topic vehicle_status for autoware.ai version 1.14?

EricBoiseLGSVL, 2021-06-29:
No sensor supports this yet afaik.  @hadiTab do we have plans to work on this or should users create their own sensor?

EricBoiseLGSVL, 2021-06-29:
Moving support to #1527 since this issue is closed