lgsvl/simulator #241: How do I create a new intersection

christian-lanius, 2019-07-16:
Hi everyone,
I am currently working on creating a new HD map for Apollo 3.5. Unfortunately, I can not find any information regarding on how to create a new intersection without traffic lights/stop signs. The only documentation I can find regarding map annotation is this link: https://www.lgsvlsimulator.com/docs/map-annotation
While going through the source code, I found a MapIntersectionBuilder class, however it is not exposed by the GUI and I am not sure how to use it.
Can you provide a simple example on how to create an intersection? As an example, consider the T intersection below
[image]

I thought about creating mapjunctionbuilder elements for the lanes I want. But then I am unsure how to link these together as a one intersection, as well as link them to the incoming maplanesegmentbuilders

EricBoiseLGSVL, 2019-07-16:
@christian-lanius 

Yes, the MapIntersectionBuilder is not in the tool.  The easiest way to look at an example is in SimpleMap.scene.  In this scene, we have intersections very similar to your example.  You can just duplicate the Intersection holder object, move it and re-parent.  I'll break down the classes and hierarchy here for you if you want to build from scratch.

MapIntersectionBuilder.cs is a component on the parent object of the intersection.  This class gets all the intersection lanes and IntersectionComponent.cs.  It also holds the queue if isStopSign is true for stop sign intersections.

IntersectionComponent.cs is a component on the first child object.  It finds the facing and opposite facing meshes for the traffic lights and controls the timing.  These are child objects with IntersectionTrafficLightSetComponent.cs.  It also creates a trigger for the queue.  It has a variable for the size of the trigger because it needs to be just inside the stoplines for that intersection.

All other annotation objects are children of MapIntersectionBuilder.cs object.  This includes the Signal, Lane and stopline annotations.

All of these aspects of an intersection are contained in the MapIntersectionBuilder parent object and are children themselves of the Intersection holder object.  This object then is placed in the public field of MapManager.cs, intersectionsHolder public variable.

The hierarchy is setup this way to separate lanes for npcs to spawn and intersections, plus handle meshes and signal annotation.

Hope this helps.

christian-lanius, 2019-07-17:
Thank you for your detailed answer. I have some follow up questions though. 
I pulled a fresh copy from the repository right now and my SimpleMap scene looks quite different to the screenshot you posted. I have followed the setup in Shalun instead and can successfully create intersections!
My intersection is not aligned in x-z, should I rotate the gameobject holding the MapIntersectionBuilder so that the bounding box is aligned to the actual crossing?
In the documentation, boundary lanes are created (the white segments in my screenshot) and it is described that the center lane should be annotated similarily. However in neither SF nor Shalun or SimpleMap these annotations exist. Are they still necessary? The documentation then goes on about linking the streets together, as in, assign left/right neighbor to each of them. However the respective properties (leftForward, rightReverse etc) are not exposed by the MapLaneSegmentBuilerComponent and the button ("Link Neighbor Lanes from Left to Right" etc) in the GUI does not exist either.

I am very grateful for your help so far!

EricBoiseLGSVL, 2019-07-17:
@christian-lanius glad to help.

1) It's odd that they are different, we might have changes that are not pushed to github yet but Shalun.scene is fine as an example.

2) MapIntersectionBuilder transform just needs to align by position, esp with the IntersectionComponent child object but rotation of this parent is not important.  *NOTE* If you copy the intersection signal light meshes, you will notice that the poles (holder objects) face Z forward to auto assign which signal lights face each other. Also, don't forget stop lines that overlap before the last waypoint in the approaching lane.   This way the lane can find if an intersection is next.

3) I think Autoware requires the boundry lines but not Apollo.  You can just copy the right or left annotation and move it into place if you want.

4) Yep, we removed the need to manually do this now.  You can comment out the NonSerialized headers to those variables and run to check that those variables are assigned correctly. Same with boundry and ids.  You will notice that intersection lanes are assigned init values, this might need to change with new versions of AD software.

closed 2020-04-07 by EricBoiseLGSVL