lgsvl/simulator #245: Speed in Apollo3.5 and Lgsvl Inconsistent

kanbei, 2019-07-18:
I ran Apollo3.5 with Lgsvl according to the direction on official website, but I found the speed of car in Apollo was not consistent with that in Lgsvl.  As a result, the car was at different positions when driving in Apollo and Lgsvl. When the car in Apollo was turnning the corner, due to inconsistent positions, the one in Lgsvl would hit the wall. 
What is the reason for this? My environment is Ubuntu16.

martins-mozeiko, 2019-07-18:
Hi.

Could you clarify what you mean by "car in Apollo"? Are you talking about Apollo Dreamview? Because Dreamview should always show same position as car in simulator - they match as long as localization module is working.

Or are you comparing our simulator with some other simulator?

daohu527, 2019-07-19:
May be this issue #244