lgsvl/simulator #246: NPC Vehicle at stopline

Felixlxc, 2019-07-22:
Hi,

   I'm building a map using the map annotation tools.  At the intersection, some NPC vehicle will be at the state "IsStopSign" or "Is Stop Light" and never trigger back thus a bunch of vehicles get stuck at the intersection. 

   I would like to know if there are tips for building  intersections and connect two lane sections so that the vehicle will stop and return to moving as normal.  Really appreciate your help!

EricBoiseLGSVL, 2019-07-23:
@Felixlxc 

Hello, let's see if I can help.  Since intersections can vary so much, this could be many things.  Let's start with the intersection trigger.  Do you have a stop sign intersection or a stop light intersection?

Can you play the simulator in Editor and share a screen shot of the Hierarchy panel and the Scene view panel (Not Game view),  Please select an NPC that is stuck.

I think the Trigger created when you play the simulator is too big.  It needs to be smaller and not overlap the stop lines in the intersection.

Felixlxc, 2019-09-13:
@EricBoiseLGSVL  Hi, I just realized that I should add the intersection component for the intersection object. I added them and now the NPC vehicles stop and go at the stop line.
 
Thanks for your comment! Bests

closed 2019-09-13 by Felixlxc