lgsvl/simulator #247: simulator not working with Unity 2018.2.4f1

Francisobiagwu, 2019-07-23:
I cloned the simulator repo as directed in git. I equally downloaded the old version of unity, 2018.2.4f1. However, the scenes don't open, instead I get a lot of compilation errors as shown in the screenshot attached. Please, doesn't anyone know what is responsible for this problem?

[image]
[image]

martins-mozeiko, 2019-07-23:
Please check that you have correctly installed [Git LFS](https://git-lfs.github.com/) before you clone repository. We use git lfs file storage to make cloning faster - as it will download files only needed in current commit, not whole history.

Francisobiagwu, 2019-07-23:
thanks.
Fixed.

closed 2019-07-23 by Francisobiagwu