lgsvl/simulator #248: GPS Simulation and inaccuracy

asatria2, 2019-07-24:
Hi, <br>

The current simulator GPS output have a tendency to jump around.
from: MapOrigin.cs:35
```c#
    public void GetLatitudeLongitude(double northing, double easting, out double latitude, out double longitude)
    {
        // MIT licensed conversion code from https://github.com/Turbo87/utm/blob/master/utm/conversion.py
```
The simulator gps lat-lon conversion is copied from https://github.com/Turbo87/utm/blob/master/utm/conversion.py <br>
This issue document the inaccuracy of the conversion https://github.com/Turbo87/utm/issues/36<br>
Is there any plan to improve the simulation of the GPS device and fix the conversion?  It will be great if the accuracy and noise of the GPS device can be definable and tunable.

martins-mozeiko, 2019-07-24:
We'll look into this. Thanks!
Do you have example coordinate values for which the calculation is imprecise?

asatria2, 2019-07-26:
```
"$GPGGA,050538.197,0.109518,N,0.092640,E,1,10,0.01,10.119778,M,0.000000,M,,*55"
"$GPGGA,050538.592,0.109382,N,0.089306,E,1,10,0.01,10.119258,M,0.000000,M,,*5B"
"$GPGGA,050538.966,0.109518,N,0.085778,E,1,10,0.01,10.118649,M,0.000000,M,,*5D"
```
The latitude jump from 0.109518 N to 0.109382 N and back again while driving straight. 
<br>Thanks.

daohu527, 2019-07-26:
Very interesting question, is it not across time zones? : )

asatria2, 2019-07-29:
Not really, the behavior shows up even without crossing North to South. Crossing it caused other problem but that's related how Autoware handling the conversion from lat-long to xyz.

closed 2020-04-07 by EricBoiseLGSVL