lgsvl/simulator #249: will there be a scenario engine in future roadmap ?

ZJLi2013, 2019-07-24:
hey, LG team,

        there already has  [NHTSA sample test cases](https://github.com/lgsvl/simulator/tree/master/Api/examples/NHTSA-sample-tests),  very helpful.  wonder as the topic name: will there be a general scenario engine in plan ?

       a few reasons for that: 

       1) usually the scenario test is not running only one time, it would multi times run, through varying some interested parameters(e.g. ego speed, ego position) in the same scenario, to catch the threshold of the system

      2) for verification purpose, can't accept the test results by running one test case for each type of road. e.g. lane following test case, should be tested on multi ODDs

      I check [scenario_runner from Carla](https://github.com/carla-simulator/scenario_runner), will this be a good start to migrate ?  

     Thanks
     David

daohu527, 2019-07-27:
Really a good idea!

daohu527, 2019-08-07:
@ZJLi2013 
I have not found any relevant parameters that can be changed in “ scenario_runner”. Can you tell me some information? In addition, does OpenSCENARIO have a benchmark? If not, it is difficult to judge whether the parameters are good or bad.

EricBoiseLGSVL, 2020-04-07:
@ZJLi2013 @daohu527 we are working on this feature and plan to release soon

closed 2020-04-07 by EricBoiseLGSVL