lgsvl/simulator #252: Parking slot for apollo open space planner

Andrea2202, 2019-07-25:
Hi All,
there is a way to add in the map Annotation tool the possibility to create the parking slots for the apollo open space planner?

Thanks in advance

cyberphysicalrobotics, 2019-07-25:
Hi Andrea,

Next version LGSVL Simulator will be released on next week. It includes parking space annotation. Then you can use apollo open space planner after annotating parking space. 

Best

Fibonacci43, 2019-07-26:
@cyberphysicalrobotics  Hi, Will LGSVL fully support apollo 5.0 in the next version? Is there a adapted apollo 5.0 repository provided with the next simulator?
Thanks in advance for your reply.

martins-mozeiko, 2019-07-26:
Yes, we will be pushing our modified Apollo 5.0 repository shortly. I believe there were not many changes and 3.5 and 5.0 codebase works exactly same with our simulator.

martins-mozeiko, 2019-08-20:
New simulator [2019.07-rc1](https://github.com/lgsvl/simulator/releases/tag/2019.07-rc1) release includes necessary annotations to run Apollo 5.0 open space planner. We've also pushed [Apollo 5.0](https://github.com/lgsvl/apollo-5.0) repository with our modifications.

closed 2020-04-07 by EricBoiseLGSVL