lgsvl/simulator #253: Why is my car model not on the ground?

yangjiongfeng, 2019-07-26:
Hello Author
I've run   ./simulator and roslaunch rosbridge_server rosbridge_websocket.launch
and then Select “XE_Rigged-autoware”
[image]
my car model not on the ground.why?
[image]

martins-mozeiko, 2019-07-26:
DuckieDowntown map is meant for small robots - like duckiebot. If you want to use this map please select duckiebot vehicle.

For real car choose a different map, for example, SanFrancisco.

yangjiongfeng, 2019-07-26:
ok 
Thank you very much.

closed 2019-07-26 by martins-mozeiko