lgsvl/simulator #254: How to replace the  car with my own car model

yangjiongfeng, 2019-07-26:
hello
 I can now use your simulation

[image]

now，i want to replace the  car with my own car model
What should I do ?
the car model file is ".xacro" or ".stl" ？

daohu527, 2019-07-26:
This issue #198 make a disscus, you can follow the step and try it out

lemketron, 2019-07-26:
The process for adding a new ego vehicle is documented here:
https://www.lgsvlsimulator.com/docs/add-new-ego-vehicle/
(and discussed in #198 as @daohu527 mentioned above)

yangjiongfeng, 2019-07-29:
@lemketron Thank you for your answer.
I follow your website tutorial
[image]
But I have some error.
[image]
I'm different from the author. What's wrong with me?
My simulator-2019.04.tar.gz is downloaded here.

daohu527, 2019-07-29:
1. Make sure your Unity version is "2018.2.4f1"
2. I'm not sure about the source code dowload [here](https://github.com/lgsvl/simulator/releases/tag/2019.04). You need to check the git lfs files are pull down. If just source code but not the lfs files, there maybe error.
3. Can you see if the console which is next to the project window has an error message?

Francisobiagwu, 2019-07-30:
What if the user wants to replace the license plate with their own license plate (maybe customized license plate)?

martins-mozeiko, 2019-07-30:
License plate currently is a texture that is applied to material. You can find texture in in this file: https://github.com/lgsvl/simulator/blob/master/Assets/Textures/CarXE/LicensePlateTexture.png

yangjiongfeng, 2019-07-31:
@daohu527 
1.What does your “You need to check the git lfs files are pull down” mean?
I can't understand
2.Can you email me your simulator_source_code document?Thank you

daohu527, 2019-07-31:
I think you can confirm the error printing in the console window first, so you can determine the problem.

You can read the git-lfs [here](https://git-lfs.github.com/), use git lfs to download because some files are too large.

yangjiongfeng, 2019-07-31:
@daohu527
Thank you very much for your answer.
I have installed git-lfs.
[image]