lgsvl/simulator #255: Apollo fork reinstall error

hyunjaecho94, 2019-07-26:
Due to a problem on my Desktop, I had to remove and reinstall Apollo (LGSVL fork). The simulator establishes connection with rosbridge, given that it runs. I can add waypoints on Dreamview and the simulator follows them. There are two issues, however. 

First, LG SVL 1080 seems to be missing from Dreamview > Tasks.

Second, under Dreamview > Module Controller, Perception toggles off itself. apollo/data/perception.out outputs `apollo/bazel-bin/modules/perception/perception: error while loading shared libraries: libcuda.so.1: cannot open shared object file: No such file or directory`. Something must have gone wrong when reinstalling, even though everything seemed to run fine when reinstalling. 

I'm not sure what libcuda.so.1 is, or how to fix this issue. These all worked fine before removing reinstalling Apollo. I followed the steps in LG SVL Apollo fork github page. 

I'm on Ubuntu 16.04.5 LTS and  GeForce GTX 1080. Thank you!

martins-mozeiko, 2019-07-26:
I assume you are running apollo inside docker we provide (dev_start/dev_into.sh scripts), right?

Did you install nvidia drivers on your host? You can check if GPU name is printed when you run `nvidia-smi`.

hyunjaecho94, 2019-07-30:
@martins-mozeiko Yes, I am running apollo inside the docker. Here's the output of` nvidia-smi`, which prints the name of the GPU. 

[image]

closed 2019-11-06 by hyunjaecho94