lgsvl/simulator #256: It's hard to understand the traffic direction

daohu527, 2019-07-30:
I think the code want to find the facing traffic lights like this, "A,C" in facingGroup and "B,D" in oppFacingGroup(Although I'm not very sure). 
[image]

I have a hard time understanding the logic of the code in "IntersectionComponent.cs  SetLightGroupData()". Why need two loops?
```
foreach (var item in lightGroups)
        {
            foreach (var group in lightGroups)
            {
                float dot = Vector3.Dot(group.transform.TransformDirection(Vector3.forward), item.transform.TransformDirection(Vector3.forward)); // TODO not vector right usually

                if (dot < -0.7f) // facing
                {
                    if (!facingGroup.Contains(item) && !oppFacingGroup.Contains(item))
                        facingGroup.Add(item);
                    if (!facingGroup.Contains(group) && !oppFacingGroup.Contains(group))
                        facingGroup.Add(group);
                }
                else if (dot > -0.5f && dot < 0.5f) // perpendicular
                {
                    if (!facingGroup.Contains(item) && !oppFacingGroup.Contains(item))
                        facingGroup.Add(item);
                    if (!oppFacingGroup.Contains(group) && !facingGroup.Contains(group))
                        oppFacingGroup.Add(group);
                }
                else if (lightGroups.Count == 1) // same direction
                {
                    if (!facingGroup.Contains(item))
                        facingGroup.Add(item);
                }
            }
        }
```

EricBoiseLGSVL, 2019-08-01:
@daohu527 

There are two loops because the method needs to find the facingGroup and the oppFacingGroup.  each element needs to be compared to all elements to find what group it needs to be assigned.  If already assigned then it is skipped.  This code handles edge cases for 1, 2, 3 or 4 way intersection light groups.

You can never rely on Unity to keep child index's the same.  If we used a simple method and assumed the order was always the same, a simple method can work.  Unfortunately, Unity can change the order of child objects each time a scene is opened, so you need a more robust method.

Hope that helps.

daohu527, 2019-08-02:
This is really a very interesting question, I need to think carefully.  
I found that trafficlight (environment - traffic light - intersection (the second)) has no stop line. Maybe this is a mistake.
My question is why I don't associate TrafficLights in "Map" with TrafficLights in the "Environment" when drawing, we might need to get a hd-map signal corresponding to the traffic lights in the game, and the current state of the traffic lights is based on The components in the "Environment". I don't know the status of the traffic light corresponding to the id.

daohu527, 2019-08-02:
I found the problem in "MapStopLineSegmentBuilder.cs"
```
           foreach (var item in mapIntersectionBuilder.intersectionC.lightGroups)
        {
            float dot = Vector3.Dot(this.transform.TransformDirection(Vector3.forward), item.transform.TransformDirection(Vector3.forward)); // TODO not vector right usually
            //if (debug) Debug.Log(dot);

            if (dot < -0.7f)
            {
                //if (debug) Debug.Log(dot);
                intersectionTrafficLightSetC = item;
                intersectionTrafficLightSetC.stopline = this;
                
            }
        }
```
some stopline can't find the intersectionTrafficLightSetC because the dot > 0.7, I find some stopline's Direction is vetor3(0,0,1) but the right is vector3(-1,0,0).

EricBoiseLGSVL, 2019-08-02:
@daohu527 

Yes, there are some issues with the intersection/traffic signal automatic assigning code.  We are about to release our new simulator and we refactored most of this.

The current setup unfortunately requires stop lines and separate holder objects for the signal light meshes to be Z forward.  This can be easy to miss and or fail with odd shaped intersections.

Be sure to remember to always put Unity objects z forward and it should reduce issues.  Since most of this code is to help automate the annotation process and can't meet all edge cases, it might be better to comment out the code, make the vars public and manually assign the needed references for difficult intersections.

Hope this helps.

daohu527, 2019-08-05:
Thanks! There is another question "SetLaneStopLines()" doesn't know if it's solved together. "lane2D" just add one point, then we can't tell if the two lines intersect.
```
var lane2D = new List<Vector2>();
lane2D.Add(lanes2D[lanes2D.Count-1]);
```
Looking forward to the new version.