lgsvl/simulator #258: rosbridge testing

kk2491, 2019-07-31:
Hi Guys,

Kindly help me in the below issue.

I am using the compiled version of LG simulator and trying to setup the ```rosbridge```. 
As given in the docs I run ```roslaunch rosbridge_server rosbridge_websocket.launch``` to start the ```rosbridge```. I could see that the simulator is now connected and able to launch the simulator successfully. 

[image]

In my local system when I run ```rostopic list``` I get the following output.

```
kk@localsys:~$ rostopic list
/apollo/sensor/camera/traffic/image_long/compressed
/client_count
/connected_clients
/lanefollowing/steering_cmd
/rosout
/rosout_agg
/simulator/control/command
/simulator/reset
/simulator/sensor/camera/center/compressed
/simulator/sensor/camera/left/compressed
/simulator/sensor/camera/right/compressed
/simulator/sensor/segmentation_camera/compressed
/simulator/sensors/lidar
```

I want to send some test messages to the control topic and see how the car is behaving. 

It seems like the topic related to the control is ```/simulator/control/command```.  But car is not moving when I try to publish test messages to that topic.

```
kk@localsys:~$ rostopic pub -r 2 /simulator/control/command geometry_msgs/TwistStamped "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
twist:
  linear:
    x: 0.5
    y: 0.0
    z: 0.0
  angular:
    x: 0.0
    y: 0.0
    z: 0.0" 
````

Could you please let me know what I am doing wrong here?

Thank you,
KK

daohu527, 2019-07-31:
"/simulator/control/command" is use by the simulator to send control commands to the outside.
I'm not sure you want to control the car in simulator? then you should choose another car like "XE_Rigged_APOLLO"

kk2491, 2019-07-31:
Thanks for the information. If I am using XE_Rigged_APOLLO then I need to run Apollo alongside the simulator. I wanted to try the simulator separately, I mean create independent ```rosbridge``` and publish the control commands rather than Apollo giving the control commands.

Is there a way I can achieve the above requirement?

Thank you,
KK

martins-mozeiko, 2019-07-31:
To control car you will need either to use different vehicle than LGSVL (Apollo or Autoware topics/messages) or add extra code to accept input from ROS. LGSVL car only accepts steering value on `/lanefollowing/steering_cmd` topic. This vehicle was created only for lane following example (see our github repo).

Also if you are using LGSVL vehicle then you need to use ROS2, not ROS1. ROS2 bridge protocol is a bit different. It probably won't work 100% correctly with ROS1 nodes. For ROS2 you need to use following bridge: https://github.com/RobotWebTools/ros2-web-bridge

Easiest way currently probably would be to use Autoware vehicle, as it uses regular ROS1 protocol. Just copy `autoware_msgs/VehicleCmd` message into your project and you'll be able to send it.

kk2491, 2019-08-02:
@martins-mozeiko Thanks for the clarification. It saved my time. 

I created the message ```autoware_msgs/VehicleCmd``` and established ```rosbridge``` by running ```roslaunch rosbridge_server rosbridge_websocket.launch```. Should I run the autoware or is it possible it send the control commands directly ? If yes, to which topic I need to publish the message ?

Thank you,
KK

martins-mozeiko, 2019-08-02:
By default autoware vehicle listens on `/vehicle_cmd` topic. Press [F1] key to see full list of topics.

Here's the code that reads the message and extracts acceleration/steering information: https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/VehicleInputController.cs#L246

kk2491, 2019-08-02:
Thats awesome. I was able to control the car by sending ros1 messages. As you suggested, I used ```autoware``` and then created ```autoware_msg``` format and published the message to ```vehicle_cmd``` topic. Unfortunately I was not knowing that there is an option [F1] to check the topics. 
Thank you so much for your support.

KK

daohu527, 2019-08-02:
good job : )

kk2491, 2019-08-02:
@martins-mozeiko @daohu527 .. Now the ```ROS1``` is done, I have started to send ```ROS2``` messages to control simulator. Initial plan was to send ```ROS2``` message then convert it to ```ROS1``` using ```ros1_bridge```. However there is issue in getting the same datatype in ```ROS2``` as in ```ROS1```. 

Is there a way I can send ```ROS2``` control messages to the Simulator directly ? If ```autoware``` itself takes ```ROS2``` messages it would be really helpful.

Thank you,
KK

daohu527, 2019-08-02:
I found autoware already support ros2. [here](https://www.autoware.auto/copy-of-home)

martins-mozeiko, 2019-08-02:
Autoware will have two versions. [Autoware.AI](https://gitlab.com/autowarefoundation/autoware.ai) which is pretty much old autoware will continue supporting ROS1. [Autoware.auto](https://gitlab.com/autowarefoundation/autoware.auto) is new rewrite of Autoware that supports only ROS2. It is maintained by Apex.AI.

Simulator supports ROS2 just fine. But if you want to use Autoware vehicle, you'll need to modify it to switch ROS type to ROS2 - each sensor has TargetEnvironment property that needs to be changed to what kind of bridge it uses (ROS1, ROS2, CyberRT).

kk2491, 2019-08-05:
@martins-mozeiko @daohu527 Apologies for the delayed response as I was out of office. 

Thanks for the information. I will definitely look into the given details and try to implement the ```ROS2``` bridge.

I was trying to send ```ROS2``` messages which goes through ```ros1_bridge```, gives the input to ```autoware``` lgsvl simulator. However there was an issue in message datatype. The message type created in ```ROS2``` is not as same as what ```autoware``` expects.  

Thank you,
KK

martins-mozeiko, 2019-08-05:
Yeah, depending on types used in message, the ROS1 bridge could sometimes send valid data to ROS2 receiving side. But this is not guaranteed. There are few ROS types that are not compatible between ROS1 and ROS2 bridges.

kk2491, 2019-08-06:
@martins-mozeiko Yea, Thanks for the info. Let me quickly check and get back to you soon.

Thank you,
KK

Amina88, 2020-03-05:
Hi guys, I have installed autoware and ros on wsl. i'm new to it does anyone know how can i see/ read the vehicle_cmd topic. Really struggling would appreciate any help. thank you

pedroexenberger, 2020-03-05:
You can run `rostopic echo vehicle_cmd` in the environment you have ROS set up.
rostopic has also other options (e.g., general topic info: `rostopic info vehicle_cmd`).

You can also see the topics info in the 'Topics' tab in Autoware. You might need to refresh the list of topics and check the 'echo' option.

Amina88, 2020-03-05:
Thanks Pedro. its giving me error unknown topic
 wsl
$ cd autoware.ai
$ rostopic echo vehicle_cmd
ERROR: Cannot load message class for [autoware_msgs/VehicleCmd]. Are your messages built?
$ rostopic info vehicle_cmd
ERROR: Unknown topic /vehicle_cmd
Any ideas how to go about fixing this. Thank you

pedroexenberger, 2020-03-05:
Are you running both autoware and lgsvl simulator, including the rosbridge setup like explained [here](https://www.lgsvlsimulator.com/docs/autoware-instructions/)?

Amina88, 2020-03-05:
Yes I am :(

Amina88, 2020-03-05:
[image]
[image]

Amina88, 2020-03-05:
Also would the autoware display like that the more i navigate through it. Any ideas what reason would be ??

pedroexenberger, 2020-03-05:
Can you check if the bridge is connected? You can check that while running the lgsvl simulator as depicted [here](https://github.com/lgsvl/simulator/issues/613#issuecomment-582047527).

Also, you may try to click on the 'Topics' tab on autoware and search for the topic you want in there. Refresh the topics to make sure you are looking to an up-to-date list of available topics. If you don't find the topic there, the bridge might not be correctly set up. 

Unfortunately, I'm not aware of why autoware display looks like this but could ask them on their forum I guess (which is in [ROS Answers](https://answers.ros.org/questions/scope:all/sort:activity-desc/tags:autoware/page:1/), using the autoware tag).

Amina88, 2020-03-05:
Thanks for helping out really appreciate it. here is what i get 
$ ./run.sh -t 1.12.0
Using options:
        ROS distro: melodic
        Image name: autoware/autoware
        Tag prefix: 1.12.0
        Cuda support: on
        Pre-release version: off
        UID: <1000>
Launching autoware/autoware:1.12.0-melodic-cuda
./run.sh: line 154: docker: command not found

Amina88, 2020-03-06:
Hi Guys, Anyone know how i can solve this issue.

£rosrun runtime_manager runtime_manager_dialog.py
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
/mnt/c/Users/Amina/autoware.ai/install/runtime_manager/lib/runtime_manager/runtime_manager_dialog.py:2865: wxPyDeprecationWarning: Call to deprecated item.
  wx.InitAllImageHandlers()

autowre still does open but when also click on LGSVL simulator this shows up and doesn't start it: Failed connect to /tmp/autoware_proc_manager
[rosrun] Couldn't find executable named rqt_lgsvl_simulator_configurator below /mnt/c/Users/Amina/autoware.ai/install/lgsvl_simulator_bridge/share/lgsvl_simulator_bridge
Failed connect to /tmp/autoware_proc_manager
Thanks for the help in advance
Unsure about it all.

pedroexenberger, 2020-03-06:
I never run LGSVL from the autoware app (perhaps there is some config that needs to be performed, but I never went after that). Instead, I navigate to the simulator folder and run `./simulator` from terminal. Check out if this works. 

Also, instead of using `rosrun` to open autoware, try `roslaunch`, because this will start various nodes, such as the master node, so it makes your life easier (you dont have to rosrun every each node on your own). You can find the roslaunch command line in the instructions page https://www.lgsvlsimulator.com/docs/autoware-instructions/#launching-autoware-alongside-lgsvl-simulator.

Amina88, 2020-03-06:
when run roslunch instead it gives me this: 
RLException: Invalid roslaunch XML syntax: not well-formed (invalid token): line 1, column 1
The traceback for the exception was written to the log file
Any idea what that could be?

Amina88, 2020-03-06:
Also when run:  rosrun runtime_manager runtime_manager_dialog.py that's what it shows. yet still opens the runtime manager yet no lgsvl or vehicle_cmd. Also what is the twist _cmd? is it same as vehicle_cmd
/mnt/c/Users/Amina/autoware.ai/install/runtime_manager/lib/runtime_manager/runtime_manager_dialog.py:2865: wxPyDeprecationWarning: Call to deprecated item.
  wx.InitAllImageHandlers()

Amina88, 2020-03-11:
$ roslaunch runtime_manager runtime_manager_dialog.py
... logging to /home/sneibus/.ros/log/ee337b6a-6376-11ea-b10d-708bcd595ab7/roslaunch-CLAYTEX33-419.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

RLException: Invalid roslaunch XML syntax: not well-formed (invalid token): line 1, column 1
The traceback for the exception was written to the log file

ANy one can help plzz

EricBoiseLGSVL, 2020-04-07:
@Amina88 can you please open a new issue?  This is an old thread that might not relate.  Also, what version of Simulator are you using and PC specs.

closed 2020-04-07 by EricBoiseLGSVL