lgsvl/simulator #259: How to get the lane line segmentation map

dwx950115, 2019-08-01:
Is there any possible to get the lane line segmentataion map use the segmentation camera?

daohu527, 2019-08-01:
Currently, lanes segmentation may not be supported, but roads segmentation is supported. The roads, buildings and sky are in different simulator layers. The simulator can directly have the ground truth. 

For now the lane is the texture of the road, so it does not seem to support lane segmentation.

EricBoiseLGSVL, 2020-04-07:
@dwx950115 yes this is not supported at this time.  We are still looking at ways to implement this while supporting road weather effects.  One way it to make invisible meshes and assign a new tag for segmentation

closed 2020-04-07 by EricBoiseLGSVL

kuku-miku, 2020-06-04:
Hi @EricBoiseLGSVL, Do we have any progress on lane segmentation using LGSVL? is it implemented? 
Thanks,

EricBoiseLGSVL, 2020-06-04:
@kuku-miku We don't support this on current open source maps but if you need to implement, create meshes for the lanes and set above the road.  They just need an alpha level of 0 in the material and a tag set for segmentation.  No need for collider.  If you want a new color, just add in SimulatorManager.prefab in editor.  Then make a new tag and assign.