lgsvl/simulator #261: Apollo 3.5 fail to run with LGSVL Simulator

jinxinyang, 2019-08-04:
I refer to [Running Apollo 3.5 with LGSVL Simulator](https://www.lgsvlsimulator.com/docs/apollo3-5-instructions/)，Most of them passed.

This is the case：
**1.Used ----./docker/scripts/dev_start.sh**：
..........
ERROR: ld.so: object 'libcaffe2_gpu.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object 'libopencv_core.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
Adding group `apollo' (GID 1000) ...
Done.
Adding user `inwinic' ...
Adding new user `inwinic' (1000) with group `apollo' ...
The home directory `/home/inwinic' already exists.  Not copying from `/etc/skel'.
[ OK ] Finished setting up Apollo docker environment. Now you can enter with: 
bash docker/scripts/dev_into.sh
[ OK ] Enjoy!
**2../docker/scripts/dev_into.sh**:
inwinic@in_dev_docker:/apollo$ 
**3. ./apollo.sh build_gpu -j 4**(If I don't add -j 4, I'll get stuck.)
..........
[INFO] Building on x86_64...
[INFO] Building with --jobs=16 --ram_utilization_factor 80 for x86_64
INFO: Reading 'startup' options from /apollo/tools/bazel.rc: --batch_cpu_scheduling --host_jvm_args=-XX:-UseParallelGC
WARNING: ignoring LD_PRELOAD in environment.
INFO: (08-04 14:07:26.998) Found 3690 targets...
INFO: (08-04 14:07:55.458) Elapsed time: 33.656s, Critical Path: 1.82s
modules/perception/proto/perception_camera.proto: warning: Import modules/common/proto/error_code.proto but not used.
INFO: Reading 'startup' options from /apollo/tools/bazel.rc: --batch_cpu_scheduling --host_jvm_args=-XX:-UseParallelGC
WARNING: ignoring LD_PRELOAD in environment.
INFO: (08-04 14:08:03.498) Found 3690 targets...
INFO: (08-04 14:08:03.605) Elapsed time: 2.453s, Critical Path: 0.00s
1============================
[ OK ] Build passed!
[INFO] Took 53 seconds
**4.bootstrap.sh**：
nohup: appending output to 'nohup.out'
nohup: appending output to 'nohup.out'
Launched module monitor.
nohup: appending output to 'nohup.out'
Launched module dreamview.
Dreamview is running at http://localhost:8888
**5.open LGSVL and Start-up simulation**
[image]
[image]
[image]
**6.open Dreamview**
First I did（
Select the XE_Rigged_Apollo3.5 vehicle and San Francisco map in the top right corner.
Open the Module Controller tap (on the left bar).
Enable Localization, Transform, Perception, Traffic Light, Planning, Prediction, Routing, and Control.）, and then I didn't respond.（There's only one big Apollo logo on the screen.，Route Editing Can't Choose）

So I can only click **sim control**
[image]
What's wrong with this? My Bridge are connected.

jinxinyang, 2019-08-05:
I used A, and then：
[image]

jinxinyang, 2019-08-05:
I used ./apollo.sh test, and then：
[image]

daohu527, 2019-08-05:
1. You don't need to open "sim_control", it is used for apollo's own simulation test.
2. I suggest you turn off "sim_control" and confirm that each module is working properly. You can refer to the [picture](https://www.lgsvlsimulator.com/docs/images/full_size_images/apollo3-5.png). If the **localization** module is working, you should be able to see the map and the car display normally.
3. Then we will look at other issues next.

jinxinyang, 2019-08-05:
I didn't choose  "sim_control", and then enabled  localization, and then（Nothing there?）
[image]
[image]

jinxinyang, 2019-08-05:
@daohu527

jinxinyang, 2019-08-05:
> 1. You don't need to open "sim_control", it is used for apollo's own simulation test.
>
>     2. I suggest you turn off "sim_control" and confirm that each module is working properly. You can refer to [...]

Can you help me remotely?
[image]

daohu527, 2019-08-05:
The localization module is ERROR. Could you check the log in "apollo/data/log" and find the error message?

jinxinyang, 2019-08-05:
> The localization module is ERROR. Could you check the log in "apollo/data/log" and find the error message?

[image]
Running build under CPU mode. No GPU is required to run the build.------Is this a mistake?

daohu527, 2019-08-05:
The GPU currently seems to have no effect on localialtion and may have an impact on perception. You can first look at the log in the "apollo/data/log" directory to confirm the error message of the localialtion module.

jinxinyang, 2019-08-05:
> The GPU currently seems to have no effect on localialtion and may have an impact on perception. You can first look at the log in the "apollo/data/log" directory to confirm the error message of the localialtion module.
[image]

If I don't select "sim_control", nothing will happen when I select "Modules" (and there are no error hints, of course), especially when I select "planning", "Routing" will be cancelled.（Others are mutually exclusive）

jinxinyang, 2019-08-05:
> The GPU currently seems to have no effect on localialtion and may have an impact on perception. You can first look at the log in the "apollo/data/log" directory to confirm the error message of the localialtion module.

    2019-08-05 19-51-57 的屏幕截图

If I don't select "sim_control", nothing will happen when I select "Modules" (and there are no error hints, of course), especially when I select "planning", "Routing" will be cancelled.（Others are mutually exclusive）

jinxinyang, 2019-08-05:
> The GPU currently seems to have no effect on localialtion and may have an impact on perception. You can first look at the log in the "apollo/data/log" directory to confirm the error message of the localialtion module.

No problem. I don't know what log to read.

daohu527, 2019-08-05:
You can use cmd "cd apollo-3.5/data/log" to enter the directory, log files are placed in this directory.

jinxinyang, 2019-08-05:
> You can use cmd "cd apollo-3.5/data/log" to enter the directory, log files are placed in this directory.
Why did I close the problem... I didn't do anything.

jinxinyang, 2019-08-05:
> You can use cmd "cd apollo-3.5/data/log" to enter the directory, log files are placed in this directory.

This is the case with log files.

[image]

martins-mozeiko, 2019-08-05:
@jinxinyang 

Velodyne module does not matter. You should NOT enable it. It is used to receive data from real hardware sensor. Which is irrelevant to our simulator. Our simulator published lidar data directly on topic other nodes are listening.

1) do NOT enable SimControl. This is not related to our simulator and is used to generate fake input data.

2) do ENABLE all required modules - **Localization, Perception, Planning, Prediction, Routing, Traffic Light, Transform**. do NOT enable other modules.

3) If any module turns off automatically, check `data/log/MODULE*.log` file. It should contain some kind of error message.

martins-mozeiko, 2019-08-07:
@jinxinyang what address do you use in Simulator for connecting to Apollo? It should be `localhost:9090` Please check this.

jinxinyang, 2019-08-08:
> @jinxinyang what address do you use in Simulator for connecting to Apollo? It should be `localhost:9090` Please check this.

According to your help to modify the simulation connection address, I appeared in  “Dreamview”  screen, and according to the tutorial selected “Localization, Transform, Perception, Traffic Light, Planning, Prediction, Routing, and Control.” (click Traffic Light and Perception, automatically cancel)。and “send routing request”（It has a red-line path.）. 
[image]
[image]
But there is another problem: after planning the path, the car will not move.
（These things are different from official tutorials.）
[image]
[image]
I also used some detection commands, which turned out to be the case.
[image]
Is there anything I lack? Or did I do something wrong?

jinxinyang, 2019-08-08:
> @jinxinyang what address do you use in Simulator for connecting to Apollo? It should be `localhost:9090` Please check this.

I also looked at "planning.log.....", indicating “Routing is not ready”.（But I'm already “send routing request”.）
[image]

daohu527, 2019-08-08:
You need to send a padmessge in apollo. When control module receive the message it will work. You can use "scripts/pad.sh" to send the message, besure to run in the apollo docker.

jinxinyang, 2019-08-08:
> You need to send a padmessge in apollo. When control module receive the message it will work. You can use "scripts/pad.sh" to send the message, besure to run in the apollo docker.

What padmessge to send, I tried. Then I started “module” first, then used “1 auto drive mode”（at pad_terminal）,Finally, I did “send routing request”.，and the result was still only red line.
[image]
[image]
[image]

[image]

jinxinyang, 2019-08-08:
I tried, too. First I did “send routing request”., then I did ‘1 auto drive mode’.
. This method is not good either.

jinxinyang, 2019-08-08:
But there's log information that didn't exist before.（IMU Related Errors,AND，Information after “I0808 13:21:45.543577 18020 class_loader_utility.h:78]registerclass:TrafficLightsPerceptionComponent,apollo::cyber::ComponentBase,/apollo/bazel-bin/modules/perception/onboard/component/libperception_component_camera.so”）
[image]
[image]

daohu527, 2019-08-08:
Your statement is very contradictory. I have seen the planned path you showed in the picture. I also saw the obstacle informations. But the cyber_monitor does not show the topics as you show in the pic.
The pad message is only valid when the planned path appears (blue wide line). If you don't have a planned path, first make sure there is "/apollo/perception/obstacles" topic receive. If the perception module and the routing module can't work, the planning module can't.
So I suggest that you sort out these ideas, not just try them, which will reduce some errors.

jinxinyang, 2019-08-08:
> Your statement is very contradictory. I have seen the planned path you showed in the picture. I also saw the obstacle informations. But the cyber_monitor does not show the topics as you show in the [...]

But there is another problem: after planning the path, the car will not move.
（These things are different from official tutorials.）：
There are official tutorials, but I don't have one.
[image]
[image]

jinxinyang, 2019-08-08:
This is my picture.
[image]

jinxinyang, 2019-08-08:
> Your statement is very contradictory. I have seen the planned path you showed in the picture. I also saw the obstacle informations. But the cyber_monitor does not show the topics as you show in the [...]
(I can't have a blue-wide line right now)

First of all, there are problems with module A and module B: but now there are no errors in their logs.
[image]
[image]
”traffic light“ There must be some problems, but I don't know where I can see the log of ”traffic light“.

martins-mozeiko, 2019-08-08:
@daohu527 This is first time I hear that you need to run `pad.sh` script to work with LGSVL simulator.
For us it works without running pad.sh script.

@jinxinyang If you are using RTX 2080 then perception and trafficlight modules won't work. Apollo only supports GTX 1080 (or similar GPUs). See my response in #269.
Car will not move because it won't be able to tell detect traffic light color (is it green or red).

jinxinyang, 2019-08-08:
Thank you. I changed the gtx1070.. I can do it,

closed 2019-08-08 by jinxinyang

daohu527, 2019-08-08:
I may confuse the pictures, I think he has a planned road.

luqiang21, 2019-08-08:
@jinxinyang For apollo 3.5, `planning` won't work if `perception`/`prediction` is not working. You can run apollo without `perception` by running fake prediction:

`mainboard -d modules/tools/prediction/fake_prediction/fake_prediction.dag`

With fake prediction, your planning will work, then after you enable control, the car should move.