lgsvl/simulator #262: LGSVL and Apollo-3.5 interfaces

Prj1508, 2019-08-05:
I am trying to understand the interfaces between LGSVL and Apollo from both the ends. I understand LGSVL subs/pubs topics from/to Apollo, but is it implemented as a typical ROS sub/pub in Simulator? What are the main changes done to base Apollo to enable this data exchange with simulator? 
I would also like to know how to add a new topic for this interface? Please let me know

daohu527, 2019-08-05:
1. You mentioned the **Apollo3.5**.
2. In fact, you need to add a bridge module in apollo3.5. lgsvl sends the message to Apollo via a TCP connection, and then the node created in apollo3.5 sends the message about the topic. 
3. If you want to add a new topic, You can directly reference the API in the game engine like this:  
```
Apollo35ChassisWriter = Bridge.AddWriter<apollo.canbus.Chassis>(ApolloTopic);
```
> you can reference "Canbus.cs"

martins-mozeiko, 2019-08-05:
@Prj1508  Apollo 3.5 replaced ROS with CyberRT. Conceptually the protocol does same things - pub/sub, topics, messages, nodes. Its just that the code is completely rewritten and API is different. The good thing is that in simulator there is almost no difference of using it - see example code daohu527 is showing above - this kind of code will work with any - ros1, ros2 or cyberRT bridge. Only difference is different topic name and message type.

Prj1508, 2019-08-05:
Thank you. 
I am just trying to monitor the topics being published by Apollo, but I cannot launch cyber_visualizer, but its throwing error as below. 

  Could not initialize GLX
  Aborted (core dumped)

Thanks for your help

martins-mozeiko, 2019-08-05:
To run cyber_visualizer you need to unset LD_PRELOAD env variable. But don't do it permanently as this variable is needed for perception module.

Prj1508, 2019-08-05:
Thank you. As I am fairly new to this, please let me know how to unset
LD_PRELOAD variable

martins-mozeiko, 2019-08-05:
Execute following line in terminal:

`unset LD_PRELOAD`

Prj1508, 2019-08-05:
thanks again. This solved it and I can bring up cyber visualizer. However, I am not able to launch two docker terminals. If I do, the  one I launched first freezes to respond
I think I need to run one docker to initiate LGSVL bridge and another to launch cyber_visualizer. Please let me know.

martins-mozeiko, 2019-08-05:
When you say "launch docker terminals" how exactly are you doing this? 
Running `docker/scripts/dev_into.sh` should work regardless of how many times you run it.
Alternative is to install tmux inside docker. And then use tmux to open multiple windows/panes for running scripts.

Prj1508, 2019-08-05:
I am running docker/scripts/dev_into.sh.
After I have one docker terminal, if I open another terminal and run
docker/scripts/dev_into.sh,
the previous docker terminal freezes.

martins-mozeiko, 2019-08-05:
Are you sure you are running dev_into.sh and not dev_start.sh second time?

Prj1508, 2019-08-05:
My bad. Thank you. Its working now

Prj1508, 2019-08-07:
I am now debugging the issue of car crashing in some route request due to frame rate misatch. I have Apollo in debug mode connected to simulator. I have these questions
1. Is there anyway I can debug two or more modules in one debug session in Apollo ( in this case, Control and planning)?
2. I would also like to know if I can debug CAN module at all in simulation mode.  How does Apollo exchange data in simulation for CAN messages? Is it via sub/pub topics thru sim and CAN messages are also updated in parallel ( though not used in sim) ?

martins-mozeiko, 2019-08-08:
1. I'm pretty sure you can run as many modules as you wish under gdb. Just open new terminal and run gdb with command you want, then execute "run" with arguments.

2. Simulator does not simulate CAN messages as we are publishing data with cyber messages. Apollo probably has some cyber node that reads/write CAN messages from Linux driver and then does all other data communications over Cyber to control car or listen for feedback. In our case simulator itself is a car, so there is no need for CAN module.

closed 2020-04-07 by EricBoiseLGSVL