lgsvl/simulator #264: Which branch should I choose for my git code?

jinxinyang, 2019-08-05:
[image]

I downloaded master-----  no bridge.sh.

Downloaded  #simulator - - ---- #261 appeared.

daohu527, 2019-08-05:
Should be the **simulator** branch, this information can be added to the "readme.md"

jinxinyang, 2019-08-05:
I should run     **./docker/scripts/dev_into.sh**     in Apollo-3.5 in home.
Or you should run **./docker/scripts/dev_into.sh**   in Apollo in /.
[image]

jinxinyang, 2019-08-05:
> Should be the **simulator** branch, this information can be added to the "readme.md"

？？？？Below

jinxinyang, 2019-08-05:
> Should be the **simulator** branch, this information can be added to the "readme.md"

Below

jinxinyang, 2019-08-05:
> Should be the **simulator** branch, this information can be added to the "readme.md"

Above

daohu527, 2019-08-05:
Assuming your code directory is “Apollo-3.5”, execute the command in Apollo-3.5.

jinxinyang, 2019-08-05:
> Assuming your code directory is “Apollo-3.5”, execute the command in Apollo-3.5.

[image]
[image]
Both places have it.

jinxinyang, 2019-08-05:
> > Assuming your code directory is “Apollo-3.5”, execute the command in Apollo-3.5.
> 
> [image]
> [image]
> Both places have it.

/apollo seems to be generated automatically. (When running ./docker/scripts/dev_start.sh)

daohu527, 2019-08-05:
Yes, you are right, they are soft links. You can view it by "ls -l"

jinxinyang, 2019-08-05:
> Yes, you are right, they are soft links. You can view it by "ls -l"

Thanks for your help

closed 2019-08-20 by martins-mozeiko