lgsvl/simulator #265: Apollo can't connect properly

jinxinyang, 2019-08-05:
> You can use cmd "cd apollo-3.5/data/log" to enter the directory, log files are placed in this directory.

This is the case with log files.

[image]

_Originally posted by @jinxinyang in https://github.com/lgsvl/simulator/issues/261#issuecomment-518207854_

jinxinyang, 2019-08-05:
Which one do I want to see?（These are all after opening “bridge.sh”.）
[image]

martins-mozeiko, 2019-08-05:
Please do not open issues with same question. Let's continue discussion in #261

closed 2019-08-05 by martins-mozeiko