lgsvl/simulator #266: dreamview.INFO have fail

jinxinyang, 2019-08-06:
[image]
[image]
[image]
What happened?

jinxinyang, 2019-08-06:
localization log is:
[image]
[image]

luqiang21, 2019-08-06:
@jinxinyang Can you elaborate more on what is your question? The fail log in dreamview.INFO might not have any effect on your simulation. As long as you can open dreamview in your browser, typically we don't have problems with dreamview. You can follow our README and enable corresponding sensors/modules on simulator/dreamview to do the simulation.

jinxinyang, 2019-08-07:
> @jinxinyang Can you elaborate more on what is your question? The fail log in dreamview.INFO might not have any effect on your simulation. As long as you can open dreamview in your browser, typically [...]

[image]
[image]
[image]
Nothing will happen.Not on the left.

jinxinyang, 2019-08-07:
> @jinxinyang Can you elaborate more on what is your question? The fail log in dreamview.INFO might not have any effect on your simulation. As long as you can open dreamview in your browser, typically [...]

1. I did it according to “https://www.lgsvlsimulator.com/docs/apollo3-5-instructions/”. Using “Nvidia Docker”, the graphics card driver is normal.

2."https://github.com/jinxinyang/my_apollo_log" This is my log information.

martins-mozeiko, 2019-08-08:
Please continue discussion in #261

closed 2019-08-08 by martins-mozeiko