lgsvl/simulator #267: timestamp synchronize between server and client

david-gwa, 2019-08-06:
is there a better way to synchronize timestamps info between server and client ? 

currently we  do e.g. 
 
```python
    while True: 
        sim.run(timelimit=0.1)
        check_npc_behavior()
 ```

daohu527, 2019-08-06:
1. What is the actual application scenario? Some methods can be asynchronous, but most methods can only be synchronized. For example, NPC travels according to the path.
2. It's like a delay trigger, I don't think it is asynchronous. 
3. In fact, you can get the time of the simulator and then process it on the client. It may only handle the delay. If the server is faster, it may not be able to make up. 

This should be discussed in the scene, but the synchronization function I think is still necessary, long-term operation will inevitably lead to out of sync.

david-gwa, 2019-08-09:
[image]

thanks for replying   @daohu527 
I found from python API doc,  looks all run in sequential.  when I asked this issue, I thought there was a way to run python and sim server in parallel, so there is a need to make sure the timestep  is consistent on both python client and sim server side.

david-gwa, 2019-08-12:
@EricBoiseLGSVL  and @martins-mozeiko   thanks for helping

closed 2019-08-20 by martins-mozeiko