lgsvl/simulator #268: Apollo.sh test error

jinxinyang, 2019-08-07:
//modules/dreamview/backend/data_collection_monitor:data_collection_monitor_test FAILED in 36.3s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/dreamview/backend/data_collection_monitor/data_collection_monitor_test/test.log
//modules/dreamview/backend/simulation_world:simulation_world_service_test FAILED in 37.6s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/dreamview/backend/simulation_world/simulation_world_service_test/test.log
//modules/map/pnc_map:cuda_util_test                                     FAILED in 35.6s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/map/pnc_map/cuda_util_test/test.log*************
//modules/map/pnc_map:pnc_map_test                                       FAILED in 0.5s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/map/pnc_map/pnc_map_test/test.log****************
//modules/map/pnc_map:route_segments_test                                FAILED in 0.3s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/map/pnc_map/route_segments_test/test.log****************
//modules/perception/camera/test:camera_app_obstacle_camera_perception_test FAILED in 41.4s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/perception/camera/test/camera_app_obstacle_camera_perception_test/test.log
***********************
//modules/perception/camera/test:camera_lib_lane_detector_darkscnn_lane_detector_test FAILED in 50.1s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/perception/camera/test/camera_lib_lane_detector_darkscnn_lane_detector_test/test.log
//modules/perception/camera/test:camera_lib_lane_detector_denseline_lane_detector_test FAILED in 46.2s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/perception/camera/test/camera_lib_lane_detector_denseline_lane_detector_test/test.log
//modules/perception/camera/test:camera_lib_lane_postprocessor_darkscnn_lane_postprocessor_test FAILED in 136.4s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/perception/camera/test/camera_lib_lane_postprocessor_darkscnn_lane_postprocessor_test/test.log
//modules/perception/camera/test:camera_lib_lane_postprocessor_denseline_lane_postprocessor_test FAILED in 34.9s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/perception/camera/test/camera_lib_lane_postprocessor_denseline_lane_postprocessor_test/test.log
//modules/perception/camera/test:camera_lib_obstacle_detector_yolo_yolo_obstacle_detector_test FAILED in 36.0s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/perception/camera/test/camera_lib_obstacle_detector_yolo_yolo_obstacle_detector_test/test.log
//modules/perception/camera/test:camera_lib_obstacle_transformer_multicue_multicue_obstacle_transformer_test FAILED in 38.8s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/perception/camera/test/camera_lib_obstacle_transformer_multicue_multicue_obstacle_transformer_test/test.log
//modules/planning/integration_tests:garage_test                         FAILED in 2.7s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/planning/integration_tests/garage_test/test.log
//modules/planning/integration_tests:sunnyvale_big_loop_test             FAILED in 2.1s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/planning/integration_tests/sunnyvale_big_loop_test/test.log
//modules/planning/integration_tests:sunnyvale_loop_test                 FAILED in 37.2s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/planning/integration_tests/sunnyvale_loop_test/test.log
//modules/planning/navi/decider:navi_decider_test                        FAILED in 36.0s
  /home/inwinic/.cache/bazel/_bazel_inwinic/540135163923dd7d5820f3ee4b306b32/execroot/apollo/bazel-out/local-dbg/testlogs/modules/planning/navi/decider/navi_decider_test/test.log

###############1##################
**********************************************
**********************************************
exec ${PAGER:-/usr/bin/less} "$0" || exit 1
-----------------------------------------------------------------------------
Running main() from gmock_main.cc
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from DataCollectionMonitorTest
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0807 05:44:33.220937  8459 global_data.cc:150] [] host ip: 192.168.1.138
W0807 05:44:33.221225  8459 environment.h:33] [] Environment variable [CYBER_PATH] not set, fallback to 
[ RUN      ] DataCollectionMonitorTest.UpdateCollectionProgress
modules/dreamview/backend/data_collection_monitor/data_collection_monitor_test.cc:149: Failure
      Expected: 30.0
      Which is: 30
To be equal to: value
      Which is: 0
[  FAILED  ] DataCollectionMonitorTest.UpdateCollectionProgress (8 ms)
[----------] 1 test from DataCollectionMonitorTest (8 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (10 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] DataCollectionMonitorTest.UpdateCollectionProgress

 1 FAILED TEST
**********************************************
**********************************************
#################2###############
**********************************************
**********************************************
exec ${PAGER:-/usr/bin/less} "$0" || exit 1
-----------------------------------------------------------------------------
Running main() from gmock_main.cc
[... 184 lines cut ...]
I0807 05:43:58.536301  8223 util.cc:83] [] 		0
I0807 05:43:58.536304  8223 util.cc:83] [] 		0
I0807 05:43:58.536307  8223 util.cc:83] [] 		0
I0807 05:43:58.536310  8223 util.cc:83] [] 		0.00416667
I0807 05:43:58.536314  8223 util.cc:83] [] 		0.00416667
I0807 05:43:58.536346  8223 yolo_obstacle_detector.cc:127] [] model_type=RTNetInt8
I0807 05:43:58.686946  8223 rt_net.cc:708] [] Erase output: detect1_ori_conf_pred
I0807 05:43:58.686980  8223 rt_net.cc:708] [] Erase output: detect2_cls_pred
I0807 05:43:58.686985  8223 rt_net.cc:708] [] Erase output: detect2_dim_pred
I0807 05:43:58.686990  8223 rt_net.cc:708] [] Erase output: detect2_loc_pred
I0807 05:43:58.686995  8223 rt_net.cc:708] [] Erase output: detect2_obj_pred
I0807 05:43:58.687000  8223 rt_net.cc:708] [] Erase output: detect2_ori_conf_pred
I0807 05:43:58.687005  8223 rt_net.cc:708] [] Erase output: detect2_ori_pred
I0807 05:43:58.687008  8223 rt_net.cc:708] [] Erase output: detect3_cls_pred
I0807 05:43:58.687013  8223 rt_net.cc:708] [] Erase output: detect3_dim_pred
I0807 05:43:58.687018  8223 rt_net.cc:708] [] Erase output: detect3_loc_pred
I0807 05:43:58.687022  8223 rt_net.cc:708] [] Erase output: detect3_obj_pred
I0807 05:43:58.687027  8223 rt_net.cc:708] [] Erase output: detect3_ori_conf_pred
I0807 05:43:58.687031  8223 rt_net.cc:708] [] Erase output: detect3_ori_pred
I0807 05:43:58.687036  8223 rt_net.cc:708] [] Erase output: lof_pred
I0807 05:43:58.687041  8223 rt_net.cc:708] [] Erase output: lor_pred
W0807 05:43:58.687319  8223 rt_net.cc:652] [] Device Not Supports Int8 Mode. Use FP32 Mode.
I0807 05:44:34.522891  8223 rt_net.cc:30] [] cudnnConvolutionLayer.cpp (254) - Cuda Error in execute: 7
I0807 05:44:34.526206  8223 rt_net.cc:30] [] cudnnConvolutionLayer.cpp (254) - Cuda Error in execute: 7
external/bazel_tools/tools/test/test-setup.sh: line 169:  8223 Segmentation fault      (core dumped) "${TEST_PATH}" "$@"

jinxinyang, 2019-08-07:
I only listed the first six questions.

jinxinyang, 2019-08-07:
#261 #266 Because of these problems, I will do this test.

martins-mozeiko, 2019-08-07:
What are the six questions?

If this is about cuda error and you are using RTX 2070 (from #269) then this is not supported by Apollo. It requires GTX 1080 (or different GPU from same generation).

jinxinyang, 2019-08-07:
> What are the six questions?
> 
> If this is about cuda error and you are using RTX 2070 (from #269) then this is not supported by Apollo. It requires GTX 1080 (or different GPU from same generation).

The above questions were all tested with "./apollo.sh test" . Now I follow Step https://www.lgsvlsimulator.com/docs/apollo3-5-instructions/#launching-apollo-alongside-the-simulator and there is still no map in it.

martins-mozeiko, 2019-08-08:
Please continue discussion in #261

closed 2019-08-08 by martins-mozeiko