lgsvl/simulator #271: API WebSocket Issue

moenasrallah, 2019-08-08:
I get the following when trying to run the python scripts for API.

Seems to be a websockets problem? Tried following the directions on multiple computers with no luck. The simulator is running and I used to use the API, but now it seems like it fails to connect?

Thank you

```
xception in thread Thread-6:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/home/Documents/simulator/Api/lgsvl/remote.py", line 28, in run
    self.loop.run_until_complete(self.process())
  File "/usr/lib/python3.5/asyncio/base_events.py", line 387, in run_until_complete
    return future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/tasks.py", line 241, in _step
    result = coro.throw(exc)
  File "/home/Documents/simulator/Api/lgsvl/remote.py", line 36, in process
    self.websocket = await websockets.connect(self.endpoint, compression=None)
  File "/home/.local/lib/python3.5/site-packages/websockets/py35/client.py", line 12, in __await_impl__
    transport, protocol = await self._creating_connection
  File "/usr/lib/python3.5/asyncio/base_events.py", line 695, in create_connection
    raise exceptions[0]
  File "/usr/lib/python3.5/asyncio/base_events.py", line 682, in create_connection
    yield from self.sock_connect(sock, address)
  File "/usr/lib/python3.5/asyncio/selector_events.py", line 402, in sock_connect
    return (yield from fut)
  File "/usr/lib/python3.5/asyncio/futures.py", line 361, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 296, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/selector_events.py", line 432, in _sock_connect_cb
    raise OSError(err, 'Connect call failed %s' % (address,))
ConnectionRefusedError: [Errno 111] Connect call failed ('127.0.0.1', 8181)
```

martins-mozeiko, 2019-08-08:
Can you try opening http://127.0.0.1:8181 in your browser? It should show http error 400 - that means websocket is listening. If it shows connection refused / cannot connect then something is not working on simulator side. Maybe you have some firewall blocking simulator executable? Also check if Linux shows that simulator is listening on port 8181 by running `ss -tlnap | grep 8181`

moenasrallah, 2019-08-13:
`ss -tlnap | grep 8181` shows that Unity is not listening on linux
also ran `sudo ufw status` and it shows firewall is inactive

Figured simulator needs to be started in the Menu scene for the connection to popup and accept API interface. Forgot about that. Works now. Thank you

Closed.

closed 2019-08-13 by moenasrallah