lgsvl/simulator #273: How Can I get the radar's data from lgsvl-sim by python script

JIXIMD, 2019-08-12:


daohu527, 2019-08-12:
Currently there is no python Api to save radar information, but there is a lidar and camera information.
I don't know if the new version is added.

JIXIMD, 2019-08-12:
Is there any other way to get radar measurements?

martins-mozeiko, 2019-08-12:
Because our radar simulation is done on high level (output is ROS/Cyber message) there is no low level data to get. If all you need is nearby cars in range of radar, you can collect them yourself by querying position of each npc and checking if their position/rotations/velocity meets your criteria about being detected.

JIXIMD, 2019-08-13:
Thank you for @daohu527 @martins-mozeiko replys. right, I want to get the relative distance and speed between NPC vehicles and EGOvehicles.My model is shown in the picture below
[image]
my python code:
     dis_x = ego.state.transform.position.x - npc.state.transform.position.x
     v_x	= ego.state.velocity.x - npc.state.velocity.x
But I get the error message :
  v_x	= ego.state.velocity.x - npc.state.velocity.x
                                                                         ^
  IndentationError: unindent does not match any outer indentation level

Then,How do I get the npc car's velocity?

closed 2019-08-13 by JIXIMD