lgsvl/simulator #275: How to Control ego car through Ros

JIXIMD, 2019-08-13:
Hi, lg team
  Now, I want to control the  throttle and brake of the ego car through my ros node without the simulator connecting Apollo or autoware.But when the simulator connects to rosbrige, all I see is that rosbrige subscribes to steer_command and publishes control_command, not that it subscribes to control_command(as below).So I would like to ask how my ros node should control the ego car.

[image]

Thank you for your help!

martins-mozeiko, 2019-08-13:
`/simulator/control/command` is used in lanefollowing example to gather training data. Simulator publishes current steering angle and acceleration which is set from steering wheel (we use Logitech G920).

`/lanefollowing/steering_cmd` is topic what simulator listens for getting only steering angle. Lanefollowing example is set up only to accept steering angle, not full control of car.

If you don't want to use Apollo or Autoware you'll need to adjust code to receive steering command in format you want.

JIXIMD, 2019-08-14:
Thank you for your reply.yeah, I don't want to use Apollo or Autoware  and I just want to implement longitudinal controls (throttle and brake) for the ego car by Ros, but I don't see a simulator subscribes to the corresponding topic.How do I do that?

martins-mozeiko, 2019-08-14:
You'll need to modify `VehicleController.cs` file to process input from ROS message you have. Check existing code to see how it processes Autoware and Apollo message types.

alexkutsan, 2020-02-11:
@JIXIMD  did you succeed within this control. I would like to do the same

nachoSO, 2020-02-17:
I'm also interested in this. Could you please shed light on this?

EricBoiseLGSVL, 2020-04-07:
@JIXIMD @alexkutsan @nachoSO We now support custom msg types.  We are also updating our Lane Following Tutorial for a future release

closed 2020-04-07 by EricBoiseLGSVL