lgsvl/simulator #276: collision with NPC

hyunjaecho94, 2019-08-13:
I've written two scripts using the Python API. In both scripts, I have one or more NPC vehicles following a set of waypoints. However, when the EGO blocks the path of an NPC, the scene generated by the first python script stops the NPC, preventing a collision with the EGO, while the other collides with the EGO. I've checked the scripts, but I can't spot why the difference is occuring. The NPC vehicles are created similarly, and the waypoint assignment is also very similar between the two scripts. Which of the two is the intended behavior for the NPCs, and what would be causing this difference?

daohu527, 2019-08-14:
When you use the waypoint to control the NPC, you will not detect the collision but follow a fixed trajectory. I will check again.

hyunjaecho94, 2019-08-14:
@daohu527 Thanks for your response. So, if the EGO is present in the trajectory, will the NPC collide or stop?

martins-mozeiko, 2019-08-14:
Currently NPCs should try to avoid collisions when they are following waypoints. But our NPCs are not smart, they are not supposed to be used as smart and fully autonomous vehicles. They are sometimes a big dumb and they hit either other vehicles or sides of road. It all depends on exact situation (positions) and randomness. 

Purpose of NPCs is just to be background actors when we will have test scenarios. At that point all smartness of NPCs will be turned off and they will blindly follow waypoints - ignoring all collisions, which means passing trhough each other, EGO vehicles or even buildings and trees. Basically you will need to set up scenario with exact movements if you want NPCs to be at exact waypoint positions and simulator will execute this scenario.

If you'll want to NPCs drive around a bit smarter then only way will be to allow them to roam randomly.

hyunjaecho94, 2019-08-15:
@martins-mozeiko Could you elaborate on "It all depends on exact situation (positions) and randomness"? I have two scripts that both makes NPCs to follow some waypoints, but the NPCs spawned by one script always does not collide when the EGO is in its way while the other collides with the EGO if it's in its way.

martins-mozeiko, 2019-08-15:
If you say you have two scripts - what is the difference between them?

Also - in current simulator version we do not guarantee deterministic execution. Different runs could produce different results depending on system performance. But we are working to have deterministic execution, we'll have it for next release (at least for NPCs).

hyunjaecho94, 2019-08-15:
@martins-mozeiko By two scripts, I mean this:

script1: Like in EOV_S_45_40.py, the NPC crashes into the EGO if the EGO does not move out of its way.

script2: Say I spawn the EGO and NPC1, which starts behind the EGO and is given waypoints to drive forward past the EGO's x position. If the EGO does not move, NPC1 will stop and not collide. 

Similarly, if NPC1 is behind NPC2 (instead of behind the EGO), and NPC1 drives forward, it stops before crashing into NPC2.

I thought maybe because in the case of EOV_S_45_40.py, the NPC's position is past the yellow line and that's why the crash occurs. So I changed its z position so that it is on the other side of the yellow line, but if I move the EGO past the yellow line so that it's in the way of the NPC, it still crashes.

EricBoiseLGSVL, 2020-04-07:
@hyunjaecho94 we are currently working on NPC refactor that will enable extensible and plug-in behavior.  This feature will release soon

closed 2020-04-07 by EricBoiseLGSVL