lgsvl/simulator #278: New version of lgsvl simulator

Fibonacci43, 2019-08-14:
Hello, is there a release plan for the next version of simulator which support apollo 5.0?

martins-mozeiko, 2019-08-14:
Apollo 5.0 is almost same as 3.5 from simulator point of view. Apollo 3.5 vehicle in current version of simulator is working with Apollo 5.0 without any changes. You should be able to run it right now. But we are currently preparing detailed step-by-step instructions with newer version of Apollo 5.0.

Fibonacci43, 2019-08-15:
@martins-mozeiko  Thank you for your reply. As I know, current simulator does not support the simulation of open space planner of Apollo. Would you add the support of this new feature in current version or next version? Is there a schedule for this feature.

martins-mozeiko, 2019-08-15:
Yes, we will support open space planner in next release (which is coming in next few days). You'll be able to run it with Apollo 5.0.

martins-mozeiko, 2019-08-20:
New simulator [2019.07-rc1](https://github.com/lgsvl/simulator/releases/tag/2019.07-rc1) release includes necessary annotations to run Apollo 5.0 open space planner. We've also pushed [Apollo 5.0](https://github.com/lgsvl/apollo-5.0) repository with our modifications.

closed 2019-08-20 by martins-mozeiko