lgsvl/simulator #279: radar occlusion characteristics

yumianhuli2, 2019-08-14:
Can your simulator reflect the occlusion characteristics of the radar?Thank U！

martins-mozeiko, 2019-08-14:
Currently radar is simulated on very high level (as ROS/Cyber message). It does that by collecting exact positions of other vehicles in the range. It does not simulate actual radar waves, reflection or occlusion.

yumianhuli2, 2019-08-15:
> Currently radar is simulated on very high level (as ROS/Cyber message). It does that by collecting exact positions of other vehicles in the range. It does not simulate actual radar waves, reflection or occlusion.

Have a plan for it？Thank U！

martins-mozeiko, 2019-08-15:
Not at the moment. Currently we are focused supporting Apollo and Autoware AD stacks. For them radar simulation on high level is enough.

Supporting more detailed simulation requires more information about requirements to know how to implement simulation. What exactly needs to be simulated, how it will be used, etc...

All the source of radar simulation is available in our github. You can always check how it is implemented and suggest how we can improve it to fit your use case.

closed 2020-04-07 by EricBoiseLGSVL