lgsvl/simulator #280: How do I translate speed and acceleration into throttle and brake to control ego car?

JIXIMD, 2019-08-15:
Hi, lg team
   Is there an API interface to support this transformation?
   If not, what should I do? 
Thank you for your help

daohu527, 2019-08-15:
What is the application scenario? There is currently no such API available. It seems that PID control can be used. If the speed is lower than expected, the throttle will be added. If it is higher than expected, the throttle will be reduced.

JIXIMD, 2019-08-15:
yes, it is a PID control。I calculate my pid output, which is an acceleration compensation.But I don't know how to map it to the accelerator and brake of the ego car

JIXIMD, 2019-08-15:
yes, it is a PID control. I calculate my pid output, which is an acceleration compensation.But I don't know how to map it to the throttle and brake of the ego car

JIXIMD, 2019-08-15:
Their mapping relationship is nonlinear, can you help to provide their mapping relationship

daohu527, 2019-08-15:
I don't know about this piece and see if others can help you.

cyberphysicalrobotics, 2019-08-15:
In apollo control module, there is control calibration table and other compensations by control options.
You can refer this fact to the following files:
/apollo/modules/control/conf/control_conf.pb.txt
/apollo/modules/control/controller/lon_controller.cc

closed 2020-04-07 by EricBoiseLGSVL