lgsvl/simulator #281: How do I get the throttle, brake, and acceleration value of the ego

JIXIMD, 2019-08-15:
Hi, lg team
   I don't see any API interfaces in v2019.05.Is there any way to get this information
Thank you for your help

daohu527, 2019-08-15:
Currently this information is given to the emulator by the control module, so the emulator does not have this information provided.
If you want to control the car with a throttle brake, you know this value, why do you need the simulator to provide it, you can check the range of values, try to give some values ​​and see if you can meet your requirements.

JIXIMD, 2019-08-15:
I want to set different speed of the ego car at different time periods and collect the these data at the same time, so as to get the approximate relationship between speed, acceleration, accelerator and brake

JIXIMD, 2019-08-15:
I want to set different speed of the ego car at different time periods and collect the these data at the same time, so as to get the approximate relationship between speed, acceleration, throttle and brake

daohu527, 2019-08-15:
You can set the position, speed of the car, and then you will get the curve. The car model is now a dynamic model and needs someone else to check if there is such a curve.
You have provided a good idea to improve the car's control method : )

martins-mozeiko, 2019-08-15:
API does not provide this information because currently in simulator you are not supposed to drive EGO car with API. API is meant for setting up scenarios / test cases where NPC/pedestrian placement matters. But EGO car itself should be controlled through ROS/Cyber messages.

closed 2020-04-07 by EricBoiseLGSVL