lgsvl/simulator #282: Wrong repository "topics"

gildor2, 2019-08-16:
Hi,

It seems like all "topics" for this repository are wrong. Examples: "unreal-engine", "airsim".
In fact it appears like ALL topics after "machine-learning" are wrong.

[image]

Thanks,
Konstantin

zelenkovsky, 2019-08-17:
Hello Konstantin,

  Thanks for your feedback, but I'm pretty sure topics of this repo are very much relevant, because people who's interested in tensorflow, artificial-intelegence and other mentioned keywords will be exposed to our repository and discover very useful autonomous vehicle simulator.

BR
Dmitry Z.

gildor2, 2019-08-17:
Hi Dmitry,

Thank you for your response.

I would not agree at least with some topics.
- "game-engine" - this topic is assigned to projects which represents implementation of some game engine. It is never assigned to a project which "could be interested to people who interested in game engines" or "project which uses game engine". If I'm clicking on that topic, I want to see results filled with custom game engines, not with projects which USES them.
- "unreal-engine" - your project is absolutely irrelevant to this area. Your project is built with Unity game engine, and adding "unreal", "quake", "cryengine" would not be good.

Thanks,
Konstantin

daohu527, 2019-08-19:
@gildor2 
I really appreciate that you are such a serious person. In fact, the advantage of tags is that it is easy to find the project, but sometimes I often add other related tags to promote my project. This doesn't seem to be a big deal, but it should be more professional for a truly mature community. I don't mean to offend, carla's label is more relevant.
The "3d", "Unreal Engine", "api" I think it's too broad or not very relevant.
Of course, I hope this community will be more standardized and professional, because of the open source spirit.
Hope all of you will have fun in this project!
> This does not represent an official point of view

closed 2019-08-19 by gildor2