lgsvl/simulator #285: Focus On Controller Simulation

askariz, 2019-08-20:
Hi @luqiang21 ,
because  of the limitation of PC I can not always run Apollo simulation with Perception and Traffic light detection . Is there someway to modify apollo stack to only Localization , Planning , Control , Routing work . Only concert about controller simulation .

luqiang21, 2019-08-20:
@askariz Yes, you can try to run fake prediction in Apollo by running following:

`mainboard -d modules/tools/prediction/fake_prediction/fake_prediction.dag`

This way you don't need `perception` and `prediction` modules running.

askariz, 2019-08-20:
@luqiang21  after I enable fake_prediction still can't  make it move . log in planning module show me that 
```
831 reference_line_provider.cc:535]  Failed to extract ss
egments from routing
E0820 12:06:13.849200   831 reference_line_provider.cc:576]  Failed to create ree
ference line from routing
E0820 12:06:13.849229   831 reference_line_provider.cc:199]  Fail to get referenn
ce line
E0820 12:06:13.899746   830 reference_line_provider.cc:535]  Failed to extract ss
egments from routing
E0820 12:06:13.899756   830 reference_line_provider.cc:576]  Failed to create ree
ference line from routing
E0820 12:06:13.899778   830 reference_line_provider.cc:199]  Fail to get referenn
ce line
```
In this way do I need to enable traffic light ?  what module do I need ？
Thank you .

askariz, 2019-08-20:
@luqiang21 
I can run the simulation right ! Thank you for your kindly help !
Best Wishes!

closed 2019-08-20 by askariz