lgsvl/simulator #286: 2019-07 version can't be used in ubuntu

yzl96, 2019-08-20:
Hi, I follow your tutorial in https://www.lgsvlsimulator.com/docs/build-instructions.
I'm in step 13: In the Projects tab, verify that the Simulator is using Unity Version 2019.1.10f1 from the dropdown

[image]

then I follow step 14 : Double-click the name of the project to launch Unity Editor
but nothing happens.I don't know why, I just follow your step , change nothing.

EricBoiseLGSVL, 2019-08-20:
@yzl96 

Hello, let me see if I can help.  I see that you are using Unity Hub 1.4.0.  The current Hub is at 2.1.0.  I'd update that first.  If you have issues after first opening the project, just close Unity and open the project a second time.  Let me know if that helps.

martins-mozeiko, 2019-08-20:
@yzl96 Please check that you have vulkan libraries installed on your Linux: `sudo apt install libvulkan1`

If that does not help, check the Editor log file at `~/.config/unity3d/Unity/Editor/Editor.log` location for any errors.

yzl96, 2019-08-21:
@EricBoiseLGSVL ,  I follow the document page's hub download url: https://forum.unity.com/threads/unity-hub-v2-0-0-release.677485/, I use ubuntu16.04, so I download linux version, but it's very strange, the downloaded hub version is 1.4, have try to use the simulator in ubuntu16.04.

yzl96, 2019-08-21:
[image]

Hi , @EricBoiseLGSVL  I update the hub, but when I press the Add button, and choose the project, then nothing happens?....

yzl96, 2019-08-21:
@martins-mozeiko , I'm in step 24 https://www.lgsvlsimulator.com/docs/build-instructions.: Open Build... in Unity: Simulator -> Build...
my build.. page is empty:

[image]

EricBoiseLGSVL, 2019-08-21:
@yzl96 
You need to clone the Environments and Vehicles repos into the External folder.  This is missing in our documentation and we are updating it soon.

[image]

martins-mozeiko, 2019-08-21:
@yzl96 We do not put environments and vehicles in our main simulator repository to save size. Maps and vehicles can get pretty big in size because of 3d assets and texture.

We have example environments available on our github: [CubeTown](https://github.com/lgsvl/CubeTown) and [ SingleLaneRoad](https://github.com/lgsvl/SingleLaneRoad). And example vehicle [Jaguar2015XE](https://github.com/lgsvl/Jaguar2015XE). We'll be updating these with more eventually.

yzl96, 2019-08-22:
@martins-mozeiko ， I put the environment as EricBoiseLGSVL, I download CubeTown  SingleLandRoad Jaguar2015XE in github by clone and download.the error is below:
Could not create asset from Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XETireAlbedo.png: File could not be read
Could not create asset from Assets/External/Environments/CubeTown/Materials/TiledWoodAlbedo.png: File could not be read
Could not create asset from Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XETireEmissive.png: File could not be read
Could not create asset from Assets/External/Environments/CubeTown/Materials/TiledWoodNormal.png: File could not be read
Could not create asset from Assets/External/Environments/SingleLaneRoad/Models/SingleLaneRoad/Materials/ShalunWhiteSolidFiveDiffuse.png: File could not be read
Could not create asset from Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XETireNormal.png: File could not be read
Could not create asset from Assets/External/Environments/CubeTown/Materials/HazardStripesAlbedo.jpg: File could not be read
Could not create asset from Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XETireMask.png: File could not be read
Could not create asset from Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XEHeadLightLowCookie.png: File could not be read
Attempting to draw without an active shader program?
UnityEditor.AssetPreviewUpdater:CreatePreviewForAsset(Object, Object[], String)
Vulkan: trying to draw without an active vertex or fragment program.
UnityEditor.AssetPreviewUpdater:CreatePreviewForAsset(Object, Object[], String)
Attempting to draw with missing bindings
UnityEditor.AssetPreviewUpdater:CreatePreviewForAsset(Object, Object[], String)
ImportFBX Errors:
Couldn't read file /home/sti/2019-07-simulator/Assets/External/Environments/SingleLaneRoad/Models/SingleLaneRoad/SingleLaneRoad.fbx.
Unexpected file type
ImportFBX Errors:
Couldn't read file /home/sti/2019-07-simulator/Assets/External/Vehicles/Jaguar2015XE/Models/Jaguar2015XEModel.fbx.
Unexpected file type

martins-mozeiko, 2019-08-22:
Do you have git-lfs installed? We use Git LFS for large binary files to save size when cloning repository.

yzl96, 2019-08-23:
It works, must git clone ranther than download zip/

closed 2019-08-23 by yzl96

peter-popov, 2019-09-03:
> Hi , @EricBoiseLGSVL I update the hub, but when I press the Add button, and choose the project, then nothing happens?....

Hi, @yzl96. I have the same issue. How did you solve it above?

gaohongfein, 2019-09-19:
Have you solve these problem？

EricBoiseLGSVL, 2019-09-19:
Anyone still having issues adding the project to Unity Hub, updating will solve it.

https://unity3d.com/hub/whats-new