lgsvl/simulator #287: some error in unity Editor, lgsvl version 2019.07

gaohongfein, 2019-08-20:
[image]
after i clone the repo and load the project, it occurs these error

martins-mozeiko, 2019-08-20:
If these errors go away after you press `Clear` button then ignore them. Sometimes Unity's HDRP package gives strange error messages which are not real errors.

gaohongfein, 2019-08-21:
[image]
this error while buid WeiUI

martins-mozeiko, 2019-08-21:
Please follow build instructions here: https://www.lgsvlsimulator.com/docs/build-instructions/

Make sure you did step 22 to install WebUI dependencies.

sriharshakunda, 2019-08-23:
@martins-mozeiko  I am having similar issues, when I tried to build the simulator Unity throws out an error "Kernel 'KSampleCopy4_1_x_8' not foundUnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)".
Web UI was built successfully.

martins-mozeiko, 2019-08-23:
@sriharshakunda This error message is harmless. Unity shows for its own code. But everything still works just fine, so simply ignore this message.

sriharshakunda, 2019-08-23:
@martins-mozeiko Now it throws me a different error. See attached screenshot. This is after clearing the errors and building the player.
[image]

martins-mozeiko, 2019-08-23:
Do you have `libvulkan1` installed on your Ubuntu? Unity Editor window should say "Vulkan" not "OpenGL 4.5" in the title.

Check the editor log file if for some reason it cannot initializ Vulkan. It is in `~/.config/unity3d/Unity/Editor/Editor.log` location.

sriharshakunda, 2019-08-23:
I do have vulcan. Where do I check in the Editor log for Vulkan Initialization. I did a quick search in the editor log file and found this.

[image]

sriharshakunda, 2019-08-23:
Quick update. Reinstalled vulkan drivers along with sudo apt-get install mesa-vulkan-drivers vulkan-utils and the Webui was built. currently running the build. Will update if successful.

martins-mozeiko, 2019-08-23:
I think your nvidia driver is not correctly installed. Nvidia provides its own vulkan driver. mesa-vulkan-drivers is for Intel and AMD GPU's.

Try reinstalling nvidia driver - the log file should show something like this:
```
Vulkan vendor=[NVIDIA] id=[10de]
Vulkan renderer=[GeForce GTX 1080] id=[1b80]
```

Felixlxc, 2019-08-29:
@martins-mozeiko  Hi, I see the 

Vulkan vendor=[NVIDIA] id=[10de]
Vulkan renderer=[GeForce GTX 1080] id=[1b80]

in my log file but still I can't build webui. 
[image]
This is my screenshot for build the webui.

sriharshakunda, 2019-08-29:
Did you do npm install in WebUI folder ?

Felixlxc, 2019-08-29:
@sriharshakunda  Yeah, I did. Did you solve the problem?

sriharshakunda, 2019-08-29:
@Felixlxc Yes, it was a problem with my Nvidia drivers. You can check if you installed nvidia drivers with nvidia-smi command. I did auto-install instead of selecting a manual driver. Also make sure you disabled secure boot so that the graphic card can be accessed.

Felixlxc, 2019-08-29:
@sriharshakunda  Hi, could you tell me what is the current version of your nvidia driver and what is the command you use for installation? Does it interupt your boot procedure? I'm currently using nvidia-384 driver and when I did update my driver, the booting process was interupt so I havn't updated for now and I haven't tried to use auto-install before.

sriharshakunda, 2019-08-29:
@Felixlxc My driver is nvidia-430. I followed [this](https://linuxconfig.org/how-to-install-the-nvidia-drivers-on-ubuntu-18-04-bionic-beaver-linux) and installed using autoinstall command. The login loop happens when your laptop actually switches from intel drivers to nvidia drivers. I had to do a fresh install and then installed nvidia drivers right away. But you can switch between drivers using [prime-select](https://www.linuxbabe.com/desktop-linux/switch-intel-nvidia-graphics-card-ubuntu) command.

closed 2020-04-07 by EricBoiseLGSVL

lemketron, 2020-04-14:
> Quick update. Reinstalled vulkan drivers along with sudo apt-get install mesa-vulkan-drivers vulkan-utils and the Webui was built. currently running the build. Will update if successful.

As @martins-mozeiko pointed out, "Nvidia provides its own vulkan driver. mesa-vulkan-drivers package is for Intel and AMD GPU's."

I didn't explicitly install "mesa-vulkan-drivers" but it somehow ended up on my Razer Blade running Ubuntu 18.04 after installing Nvidia drivers and libvulkan1. So even though Nvidia drivers were  installed and working fine (like for glxgears), the Simulator crashed on launch and the Player.log file indicated that Vulkan was not using the Nvidia renderer:
`Vulkan vendor=[Intel] id=[8086]`
`Vulkan renderer=[Intel(R) UHD Graphics 630 (Coffeelake 3x8 GT2)] id=[3e9b]`

The solution for me turned out to be uninstalling the mesa-vulkan-drivers:
`sudo apt-get remove mesa-vulkan-drivers`