lgsvl/simulator #288: 2019-07 version executable can't run in ubuntu16.04

yzl96, 2019-08-20:
Hi , I follow your step to use the new version simulator.
First , I enter the web page:
[image]

then, I press the start button:
[image]

then, then simulator just return the start page, 
[image]

and the output log don't change, the log is just the same as start.

log:
Set current directory to /home/sti/lgsvlsimulator-linux64-2019.07-rc1
Found path: /home/sti/lgsvlsimulator-linux64-2019.07-rc1/simulator
Mono path[0] = '/home/sti/lgsvlsimulator-linux64-2019.07-rc1/simulator_Data/Managed'
Mono config path = '/home/sti/lgsvlsimulator-linux64-2019.07-rc1/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Preloaded 'ScreenSelector.so'
Display 0 '24B1W1 24"': 1920x1080 (primary device).
Loading player data from /home/sti/lgsvlsimulator-linux64-2019.07-rc1/simulator_Data/data.unity3d
Logging to /home/sti/.config/unity3d/Editor/Player.log
[12045:12045:0820/170551.554853:ERROR:zygote_host_impl_linux.cc(89)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.
[12055:12055:0820/170551.588108:ERROR:zygote_host_impl_linux.cc(89)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.
Running Firefox as root in a regular user's session is not supported.  ($HOME is /home/sti which is owned by sti.)
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: iceweasel: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: seamonkey: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: mozilla: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: epiphany: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: konqueror: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: chromium-browser: not found
[12083:12083:0820/170551.646362:ERROR:zygote_host_impl_linux.cc(89)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: www-browser: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: links2: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: elinks: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: links: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: lynx: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: w3m: not found
xdg-open: no method available for opening 'http://localhost:8080/'

env:
ubuntu16.04
titan x

martins-mozeiko, 2019-08-20:
New Unity HDRP rendering requires Vulkan to be installed on Linux. Please check that you have vulkan library installed: `sudo apt install libvulkan`

If that does not help check the log file at `~/.config/unity3d/Editor/Player.log` location.

hadiTab, 2019-08-20:
@yzl96 I believe this happens when an address for the rosbridge is not entered. See the image below where I've entered `localhost:9090` next to the vehicle name. I believe this should fix the problem.

[image]

martins-mozeiko, 2019-08-20:
And if what Hadi is saying does not help check the log file in `~/.config/unity3d/Editor/Player.log` location for any error messages.

yzl96, 2019-08-21:
@martins-mozeiko , thanks for you , it does works, I think maybe you should add this to your document, people usually don't fill the option

closed 2019-08-21 by yzl96