lgsvl/simulator #289: API cannot connected

yoyouwen, 2019-08-20:
Hi, I am using the simulator in Windows and apollo 3.0 in another linux server. I start the simulator and run a example to make sure the car can move successfully. Then I try to use the python API but I failed: 
`Exception in thread Thread-1:                                                                                           Traceback (most recent call last):                                                                                        File "C:\Users\AppData\Local\Programs\Python\Python37\lib\threading.py", line 926, in _bootstrap_inner                self.run()                                                                                                            File "D:\ChromeDownloads\simulator-2019.05\simulator-2019.05\Api\lgsvl\remote.py", line 28, in run                        self.loop.run_until_complete(self.process())                                                                          File "C:\Users\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 579, in run_until_complete                                                                                                                            return future.result()                                                                                                File "D:\ChromeDownloads\simulator-2019.05\simulator-2019.05\Api\lgsvl\remote.py", line 36, in process                    self.websocket = await websockets.connect(self.endpoint, compression=None)                                            File "C:\Users\AppData\Local\Programs\Python\Python37\lib\site-packages\websockets\client.py", line 537, in __await_impl__                                                                                                                    transport, protocol = await self._create_connection()                                                                 File "C:\Users\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 954, in create_connection     raise exceptions[0]                                                                                                   File "C:\Users\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 941, in create_connection     await self.sock_connect(sock, address)                                                                                File "C:\Users\AppData\Local\Programs\Python\Python37\lib\asyncio\selector_events.py", line 464, in sock_connect      return await fut                                                                                                      File "C:\Users\AppData\Local\Programs\Python\Python37\lib\asyncio\selector_events.py", line 494, in _sock_connect_cb                                                                                                                          raise OSError(err, f'Connect call failed {address}')                                                                ConnectionRefusedError: [Errno 10061] Connect call failed ('127.0.0.1', 8181)                    `

Then I tried opening http://127.0.0.1:8181 in browser. It shows connection refused. Is there any way to fix this? Thank you so much!

hadiTab, 2019-08-20:
@yoyouwen is 127.0.0.1 the correct address for the machine running the simulator? If not you need to enter the correct address when you create a `simulator` object in your python code. i.e.:
```
sim = lgsvl.Simulator(address = "192.168.1.2", port = 8181)
```
Replace 192.168.1.2 with the address of the simulator machine.

yoyouwen, 2019-08-20:
> @yoyouwen is 127.0.0.1 the correct address for the machine running the simulator? If not you need to enter the correct address when you create a `simulator` object in your python code. i.e.:
> 
> ```
> sim = lgsvl.Simulator(address = "192.168.1.2", port = 8181)
> ```
> 
> Replace 192.168.1.2 with the address of the simulator machine.
> 
> You should also edit the config file located in the folder where the simulator executable resides (see [documentation](https://www.lgsvlsimulator.com/docs/config-and-cmd-line-params/) on config file). `api_hostname` is set to `localhost` by default. You will need to change it to match the address of the machine you are running the API on, or you can set it to `*` to accept API calls from anywhere.

Thank you for your reply. I run the simulator and the python script on the same machine. And in the config file, I leave the  api_hostname as localhost. I try to set it to * but it does not work too.

hadiTab, 2019-08-20:
@yoyouwen I think I misunderstood your original issue. If you are running the simulator and the python scripts on the same windows machine than everything should stay as `localhost`. The linux server that is running Apollo 3.0 won't matter here (you can't run all three -- either simulator + API or simulator + Apollo).

Can you check your firewall and make sure it is not blocking the simulator?
Also which release version of the simulator are you using?

yoyouwen, 2019-08-20:
> @yoyouwen I think I misunderstood your original issue. If you are running the simulator and the python scripts on the same windows machine than everything should stay as `localhost`. The linux [...]

I am using the latest version: windows64 2019.07

hadiTab, 2019-08-20:
@yoyouwen how are you configuring your simulation in the web interface?
To use API in the latest version of the simulator you will need to select `API Only` when creating a new simulation (or editing an existing one). See image below:
[image]

yoyouwen, 2019-08-20:
> @yoyouwen how are you configuring your simulation in the web interface?
> To use API in the latest version of the simulator you will need to select `API Only` when creating a new simulation (or editing an existing one). See image below:
> [image]

I try this way. The API seems work but when I try the quickstart, I get new error:
`D:\ChromeDownloads\simulator-2019.05\simulator-2019.05\Api>python 02-loading-scene-show-spawns.py                       Current Scene = None                                                                                                    Traceback (most recent call last):                                                                                        File "02-loading-scene-show-spawns.py", line 19, in <module>                                                              sim.load("SanFrancisco")                                                                                              File "D:\ChromeDownloads\simulator-2019.05\simulator-2019.05\Api\lgsvl\utils.py", line 21, in new_f                       return f(*args, **kwargs)                                                                                             File "D:\ChromeDownloads\simulator-2019.05\simulator-2019.05\Api\lgsvl\simulator.py", line 35, in load                    self.remote.command("simulator/load_scene", {"scene": scene})                                                         File "D:\ChromeDownloads\simulator-2019.05\simulator-2019.05\Api\lgsvl\remote.py", line 68, in command                    raise Exception(data["error"])                                                                                      Exception: Environment 'SanFrancisco' is not available                                                                                                             `

hadiTab, 2019-08-20:
@yoyouwen you seem to be using the old python API scripts from a previous release (your folder name is simulator-2019.05). It is trying to load the 'SanFrancisco' scene which is not part of the new simulator release.
If you download the latest version of the python API, the quickstart scripts will have been modified to load BorregasAve instead of SanFrancisco. There may also be other minor differences as well.

yoyouwen, 2019-08-21:
> @yoyouwen you seem to be using the old python API scripts from a previous release (your folder name is simulator-2019.05). It is trying to load the `SanFrancisco` scene which is not part of the new [...]

I change the simulator to version 5 and it works! Thank you so much for your reply!

closed 2019-08-21 by martins-mozeiko