lgsvl/simulator #290: GPS sensor return non sensible value

LiyouZhou, 2019-08-20:
GPS sensor return non-sensible value.
I wrote a script to compare the gps sensor reading `gps_sensor.data`, with the Transform in the map coordinates translated to gps coordinates `sim.map_to_gps(state.transform)`. They yield vastly different result. Because I am loading the san fransisco map, the map coordinate transform seem to be correct but the gps data is not sensible.

Running the following script:

```
import os
import lgsvl
import random
import time

sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "127.0.0.1"), 8181)
if sim.current_scene == "SanFrancisco":
  sim.reset()
else:
  sim.load("SanFrancisco")

spawns = sim.get_spawn()

state = lgsvl.AgentState()
state.transform = spawns[0]
a = sim.add_agent("XE_Rigged-autoware", lgsvl.AgentType.EGO, state)

enabled_sensors = ["velodyne", "GPS", "Main Camera", "IMU"]

print(a.get_sensors())
for sensor in a.get_sensors():
	print(sensor.name, sensor.enabled)
	if sensor.name in enabled_sensors:
		sensor.enabled = True

	if sensor.name == "GPS":
		gps_sensor = sensor

gps_data = gps_sensor.data
print(gps_sensor.data)
print(sim.map_to_gps(state.transform))

sim.run(50)

gps_data = gps_sensor.data
print(gps_sensor.data)
print(sim.map_to_gps(state.transform))
```

Shows the following output:

```
GpsData(latitude=0.00178910628236925, longitude=0.00189371131328642, northing=197.75, easting=500210.72265625, altitude=10.1000003814697, orientation=-224.649066925049)
GpsData(latitude=37.7908081474212, longitude=-122.399389820989, northing=4182775.01028442, easting=52881.6509428024, altitude=10.1000003814697, orientation=-224.649066925049)
GpsData(latitude=0.00178910628236925, longitude=0.00189371131328642, northing=197.75, easting=500210.72265625, altitude=10.1244268417358, orientation=-224.649066925049)
GpsData(latitude=37.7908081474212, longitude=-122.399389820989, northing=4182775.01028442, easting=52881.6509428024, altitude=10.1000003814697, orientation=-224.649066925049)
```

martins-mozeiko, 2019-08-20:
Autoware code (even 1.12 release) by default does not support GPS coordinates outside Japan. We did a quick patch in Autoware code for it to support coordinates around (0, 0) for latitude & longitude. See https://github.com/lgsvl/Autoware/commit/b2c64116f74b30f6ca0ec43fff39afd262576c70 commit. That's why our simulator GPS sensor for Autoware vehicle is giving coordinates around 0,0 location.

If you want to see more real latitude/longitude values you need to use Apollo vehicle.

LiyouZhou, 2019-08-27:
thank you for the explanation.

closed 2019-08-27 by LiyouZhou