lgsvl/simulator #291: 2019-07 version specify a Sensor

yzl96, 2019-08-21:
Hi, I want to display different sensor's effect, like lidar, so I look the tutorial: https://www.lgsvlsimulator.com/docs/sensor-json-options/#how-to-specifiy-a-sensor
,I'm a little confused with how to use it, should I install apollo or autoware so that i can change the simulator's sensor? where is the json I can change, or it is created by myself, if it is created by me , where I should place the json?

martins-mozeiko, 2019-08-21:
Sensor configuration currently is available as raw json on configuration of vehicle. You edit configuration by pressing little wrench icon on Vehciles page:

[image]

We will be working on having better web interface to directly modify each parameter without need to edit json.

yzl96, 2019-08-21:
@martins-mozeiko  below are my config page, I see the json has lots of sensor config, and I don't see any shortcuts for display lidar effect in simulator.
[image]

[image]

yzl96, 2019-08-21:
@martins-mozeiko , I just want to use like old version, in old version , in simulator, we can choose display what, like 3d ground truth, lidar, camera and so on, you just need to click one time, but in the new version, I can see nothing, I don't want to play game in the simulator, I want to use the sensors, so I just want to if I can do like old version, choose display different things. you say I should edit the vehicle's json to configure different sensors, I try, I delete the "Manual Control" , it does work, but for sensors, the default json have all sensors, I can't use any tensor in simulator.
what's more,  in json, for each sensor's config, the Topic option is not null, will my understanding, the sensor with push msgs to the topic, and then display in simulator, I use command " rostopic hz xxx" to see if the topic is running, the result is not.

martins-mozeiko, 2019-08-21:
We are currently working on interface to have toggle button for every sensor to enable/disable visualization.

For lidar currently you can enable its visualization by adding `"Visuallize": true` in its parameters.

Ground truth sensors should visualize detected objects by default. So you simply put sensor in the list and it will show 3d bounding boxes. Default json does not have all sensors, only ones that are needed by Apollo or Autoware. This excludes ground truth sensors.

Sensor visualization does work without bridge. If there is no bridge data stays only in simulator. Bridge is used only to send data to ROS/Cyber to be used in actual AD stack.

If you don't see messages on topic that means bridge is not connected. Check your IP address & port on that is assigned to vehicle on simulation page.

Rviz does work with sensor data sent to ROS bridge (Autoware vehicle).

yzl96, 2019-08-22:
@martins-mozeiko , this is my setting of lidar:
{"type": "Lidar", "name": "Lidar",
            "params": {"LaserCount": 32, "MinDistance": 0.5, "MaxDistance": 100, "RotationFrequency": 10, "MeasurementsPerRotation": 360, "FieldOfView": 41.33, "CenterAngle": 10, "Compensated": true, "PointColor": "#ff000000", "Topic": "/points_raw", "Frame": "velodyne", "Visuallize": true},
            "transform": {"x": 0, "y": 2.312, "z": -0.3679201, "pitch": 0, "yaw": 0, "roll": 0}},
after I add "Visuallize": true, the simulator can't run, it have the same problem I occur before, push the start button, and then the simulator return to open browser page, I can't use the simulator, and then I delete "Visuallize": true, the simulator can work.

martins-mozeiko, 2019-08-22:
Sorry, I mistyped. It is "Visualize", see here in code: https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Sensors/LidarSensor.cs#L73

yzl96, 2019-08-22:
@martins-mozeiko  it works, thanks.

yzl96, 2019-08-22:
@martins-mozeiko , for ground truth visualization, I suggest you say in document, add
 {
    "type": "3D Ground Truth",
    "name": "3D Ground Truth",
    "params": {
      "Frequency": 10,
      "Topic": "/simulator/ground_truth/3d_detections"
    },
    "transform": {
      "x": 0,
      "y": 1.975314,
      "z": -0.3679201,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
and 
{
    "type": "3D Ground Truth Visualizer",
    "name": "3D Ground Truth Visualizer",
    "params": {
      "Topic": "/simulator/ground_truth/3d_visualize"
    },
    "transform": {
      "x": 0,
      "y": 1.975314,
      "z": -0.3679201,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
}  , 
if just add "3D Ground Truth Visualizer", the simulator doesn't work

yzl96, 2019-08-22:
@martins-mozeiko , I want to display lidar in rviz, My simulator config's vehicle port is localhost:9090, I don't konw how to use ros bridge,I konw can display point cloud in simulator, i want to see point cloud in rviz, my understanding is : the vehicle's config page has sensor config, sensor config has ros topic, so I just open rviz, select add , and find the topic, then display , but now I can't find the topic in rviz.

yzl96, 2019-08-22:
@martins-mozeiko , this is the topic in rviz, I want to display 3d ground truth, in simulator, now I can see 3d ground truth, it ros topic is /simulator/ground_truth/3d_visualize, but I don't find the topic in rviz.
[image]

martins-mozeiko, 2019-08-22:
You need only "3D Ground Truth" sensor.

"3D** Ground Truth Visualizer" is needed only to receive information from ROS for visualization. If you AD stack does perception then you can use this "sensor" to visualize your detected objects. This is explained in our documentation: https://www.lgsvlsimulator.com/docs/sensor-json-options/

Rviz won't visualize custom messages. 3D and 2D ground truth sensors publish custom messages from https://github.com/lgsvl/lgsvl_msgs

You can write script that will convert these messages to whatever format rviz expects. Check out #153 issue for example.

closed 2019-08-23 by yzl96